RobotLocomotion/drake #11378: ImportError: No module named vtk.

54WC, 2019-05-03:
I invoke bazel to run three version of command for drake_visualizer, all result in the same bug:ImportError: No module named vtk.  In this case, I can't append "--spdlog_level=trace" to get more information. 

(1)MacBook-Pro:drake wc$ ./bazel-bin/tools/drake_visualizer
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/mainwindowapp.py", line 2, in <module>
    from director.componentgraph import ComponentFactory
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/componentgraph.py", line 2, in <module>
    from director.fieldcontainer import FieldContainer
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/fieldcontainer.py", line 1, in <module>
    import vtk
ImportError: No module named vtk

(2)MacBook-Pro:drake wc$ bazel-bin/tools/drake_visualizer 
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/mainwindowapp.py", line 2, in <module>
    from director.componentgraph import ComponentFactory
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/componentgraph.py", line 2, in <module>
    from director.fieldcontainer import FieldContainer
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/fieldcontainer.py", line 1, in <module>
    import vtk
ImportError: No module named vtk

(3) MacBook-Pro:drake wc$ bazel run //tools:drake_visualizer 
INFO: Analysed target //tools:drake_visualizer (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //tools:drake_visualizer up-to-date:
  bazel-bin/tools/drake_visualizer
INFO: Elapsed time: 0.599s, Critical Path: 0.02s
INFO: 0 processes.
INFO: Build completed successfully, 1 total action
INFO: Build completed successfully, 1 total action
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/mainwindowapp.py", line 2, in <module>
    from director.componentgraph import ComponentFactory
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/componentgraph.py", line 2, in <module>
    from director.fieldcontainer import FieldContainer
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/fieldcontainer.py", line 1, in <module>
    import vtk
ImportError: No module named vtk

_Originally posted by @54WC in https://github.com/RobotLocomotion/drake/issues/11373#issuecomment-489150871_

jamiesnape, 2019-05-03:
What is the output of `brew linkage vtk@8.2`?

54WC, 2019-05-03:
Here is the output of brew linkage vtk@8.2

Weide-MacBook-Pro:drake wc$ brew linkage vtk@8.2
System libraries:
  /System/Library/Frameworks/AppKit.framework/Versions/C/AppKit
  /System/Library/Frameworks/Cocoa.framework/Versions/A/Cocoa
  /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
  /System/Library/Frameworks/CoreGraphics.framework/Versions/A/CoreGraphics
  /System/Library/Frameworks/Foundation.framework/Versions/C/Foundation
  /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL
  /usr/lib/libSystem.B.dylib
  /usr/lib/libc++.1.dylib
  /usr/lib/libexpat.1.dylib
  /usr/lib/libobjc.A.dylib
  /usr/lib/libxml2.2.dylib
  /usr/lib/libz.1.dylib
Homebrew libraries:
  /usr/local/opt/freetype/lib/libfreetype.6.dylib (freetype)
  /usr/local/opt/glew/lib/libGLEW.2.1.dylib (glew)
  /usr/local/opt/hdf5/lib/libhdf5.103.dylib (hdf5)
  /usr/local/opt/hdf5/lib/libhdf5_hl.100.dylib (hdf5)
  /usr/local/opt/jpeg/lib/libjpeg.9.dylib (jpeg)
  /usr/local/opt/jsoncpp/lib/libjsoncpp.19.dylib (jsoncpp)
  /usr/local/opt/libogg/lib/libogg.0.dylib (libogg)
  /usr/local/opt/libpng/lib/libpng16.16.dylib (libpng)
  /usr/local/opt/libtiff/lib/libtiff.5.dylib (libtiff)
  /usr/local/opt/lz4/lib/liblz4.1.dylib (lz4)
  /usr/local/opt/netcdf/lib/libnetcdf.15.dylib (netcdf)
  /usr/local/opt/qt/lib/QtCore.framework/Versions/5/QtCore (qt)
  /usr/local/opt/qt/lib/QtGui.framework/Versions/5/QtGui (qt)
  /usr/local/opt/qt/lib/QtSql.framework/Versions/5/QtSql (qt)
  /usr/local/opt/qt/lib/QtWidgets.framework/Versions/5/QtWidgets (qt)
  /usr/local/opt/ospray@1.8/lib/libospray.0.dylib (robotlocomotion/director/ospray@1.8)
  /usr/local/opt/ospray@1.8/lib/libospray_common.0.dylib (robotlocomotion/director/ospray@1.8)
  /usr/local/opt/vtk@8.2/lib/libvtkChartsCore-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkChartsCorePython27D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkChartsCorePython37D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonColor-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonColorPython27D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonColorPython37D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonComputationalGeometry-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonComputationalGeometryPython27D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonComputationalGeometryPython37D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonCore-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonCorePython27D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonCorePython37D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonDataModel-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonDataModelPython27D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonDataModelPython37D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonExecutionModel-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonExecutionModelPython27D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonExecutionModelPython37D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonMath-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonMathPython27D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonMathPython37D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonMisc-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonMiscPython27D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonMiscPython37D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonSystem-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonSystemPython27D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonSystemPython37D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonTransforms-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonTransformsPython27D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkCommonTransformsPython37D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkDICOMParser-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkDomainsChemistry-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkDomainsChemistryOpenGL2-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkDomainsChemistryOpenGL2Python27D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkDomainsChemistryOpenGL2Python37D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
  /usr/local/opt/vtk@8.2/lib/libvtkDomainsChemistryPython27D-8.2.1.dylib (robotlocomotion/director/vtk@8.2)
[... 275 lines cut ...]
[... cut ...]
    from director.componentgraph import ComponentFactory
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/componentgraph.py", line 2, in <module>
    from director.fieldcontainer import FieldContainer
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/fieldcontainer.py", line 1, in <module>
    import vtk
ImportError: No module named vtk

54WC, 2019-05-03:
It look like this issue is similar as #6972. I try brew link --force vtk@8.2

The result is like this: 
MacBook-Pro:drake wc$ bazel run //tools:drake_visualizer
INFO: Analysed target //tools:drake_visualizer (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //tools:drake_visualizer up-to-date:
  bazel-bin/tools/drake_visualizer
INFO: Elapsed time: 0.555s, Critical Path: 0.00s
INFO: 0 processes.
INFO: Build completed successfully, 1 total action
INFO: Build completed successfully, 1 total action
Segmentation fault: 11

Then I try : 
brew update && brew upgrade && brew cleanup
./setup/Mac/install_prereqs.sh
MacBook-Pro:drake wc$ bazel run //tools:drake_visualizer

The result like this: 
INFO: Analysed target //tools:drake_visualizer (0 packages loaded, 0 targets configured).
INFO: Found 1 target...
Target //tools:drake_visualizer up-to-date:
  bazel-bin/tools/drake_visualizer
INFO: Elapsed time: 0.871s, Critical Path: 0.00s
INFO: 0 processes.
INFO: Build completed successfully, 1 total action
INFO: Build completed successfully, 1 total action
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/drakevisualizerapp.py", line 3, in <module>
    from director import mainwindowapp
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/mainwindowapp.py", line 2, in <module>
    from director.componentgraph import ComponentFactory
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/componentgraph.py", line 2, in <module>
    from director.fieldcontainer import FieldContainer
  File "/private/var/tmp/_bazel_wc/ecfb0aaf3c5988894c5be550fe9d5e24/external/drake_visualizer/lib/python2.7/site-packages/director/fieldcontainer.py", line 1, in <module>
    import vtk
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/vtk.py", line 12, in <module>
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/importlib/__init__.py", line 37, in import_module
    __import__(name)
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/vtkmodules/all.py", line 7, in <module>
  File "/usr/local/Cellar/python@2/2.7.16/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/vtkmodules/vtkCommonCore.py", line 9, in <module>
ImportError: No module named vtkCommonCorePython

I didn't fix anything.

jamiesnape, 2019-07-22:
Fixed by re-building the `vtk@8.2` bottles. `brew reinstall vtk@8.2` would ensure the new bottles are installed.

closed 2019-07-22 by jamiesnape