RobotLocomotion/drake #11380: Obtain Python runtime from a toolchain rather than flag

jamiesnape, 2019-05-03:
Advance warning that #7899 will break Drake badly. WIP at https://github.com/jamiesnape/drake/tree/incompatible-use-python-toolchains. FYI @EricCousineau-TRI.

jamiesnape, 2019-05-03:
(Maybe we can stop having to define `DRAKE_PYTHON_BIN_PATH` too.)

EricCousineau-TRI, 2019-05-03:
Ooh, yes, removing that env var would be awesome!

EricCousineau-TRI, 2019-05-09:
Per f2f, seems that this got reverted in 0.26 'cause it broke too many things.
Will come in later versions (probably 0.27 rather).

jamiesnape, 2019-05-09:
> Per f2f, seems that this got reverted in 0.26 'cause it broke too many things.
Will come in later versions (probably 0.27 rather).

Not reverted, strictly speaking, the flip of the incompatible flag was delayed indefinitely.

jamiesnape, 2019-06-20:
I am leaning toward trying to ride out have to support selectable Python 2 and 3 support with `--incompatible` flags. Building Python 2 and 3 simultaneously would actually be easier, but we don't support Python 3 on Xenial. Supporting just Python 2 on Xenial and just Python 3 on Bionic and Mac seems easy enough, so waiting for Xenial to disappear is not an issue.

EricCousineau-TRI, 2019-11-02:
We should target to have this done by Dec. (ideally), in preparation for Bazel 2.0.

PR #12377 (2019-12-10): Define and use a toolchain for Python and use rules_python for build rules
Relates #12376 and closes #11380.

The following will need updating:

* https://github.com/RobotLocomotion/drake-external-examples/blob/master/drake_bazel_external/.bazelrc#L30-L35 - delete

* https://github.com/RobotLocomotion/drake-external-examples/blob/master/drake_bazel_external/WORKSPACE#L79-L80 - change to
```
load("@drake//tools/workspace:default.bzl", "add_default_workspace")
add_default_workspace()
```

* https://github.com/RobotLocomotion/drake-external-examples/blob/master/drake_bazel_external/apps/BUILD.baz [... cut]
closed 2019-12-10 by jamiesnape