RobotLocomotion/drake #11389: Forward dynamics for MBP

amcastro-tri, 2019-05-06:
@bobbyluig did an excellent job implementing an O(n) forward dynamics in #8194 that solves for vdot given the state and a known set of external forces.
Some work is left however in order to merge #8194 in a way that would allow us to incorporate it into our contact solvers. Currently the biggest issue in #8194 is the coupling of inverse/forward dynamics computations which prevents us from using caching effectively.

What we need is an O(n) "operator form" of the mass matrix inverse. I already have a working dev branch and I am cleaning up for PR. 

Forward dynamics should improve both performance of our current sims as well as scalability (meaning we could run sims with more objects with a cost that increases linearly in the number of dofs).

amcastro-tri, 2019-05-20:
I have code for this in my dev branch and unit tests. Working on intern projects slowed me down but definitely this is a feature we want in.

sherm1, 2020-05-12:
@amcastro-tri can we close this now?

amcastro-tri, 2020-05-13:
No operator form yet, but we do have O(n) forward dynamics. We can open an operator form issue when it becomes relevant.

closed 2020-05-13 by amcastro-tri

sherm1, 2020-05-13:
We do have an issue already for operators: #12210