RobotLocomotion/drake #11392: Incomplete debug information for ASan for Macs

m-chaturvedi, 2019-05-06:
There's a lack of line numbers in the stack for ASan at least in [`mac-mojave-clang-bazel-experimental-address-sanitizer`](https://drake-jenkins.csail.mit.edu/view/Mac%20Mojave/job/mac-mojave-clang-bazel-experimental-address-sanitizer/2/) builds, for example, [this](https://gist.github.com/m-chaturvedi/e5d242affba4c4dc57ff22591463fb67).

jamiesnape, 2019-05-20:
#11489 is at least one example of this slowing down an important feature, so I think medium is probably a fairer priority, especially given the fix is probably quicker than the time it would add next time someone has to chase around the source code for a memory issue.

EricCousineau-TRI, 2019-06-27:
Resolved by #11730

closed 2019-06-27 by EricCousineau-TRI