RobotLocomotion/drake #11403: systems: Why do certain systems do `DoCalcNextUpdateTime` adding eps and others do not?

EricCousineau-TRI, 2019-05-06:
Related to #11402, per f2f with @pangtao22:

For `DoCalcNextUpdateTime`, some systems use (a) `t_next = t_now`, while others do (b) `t_next = t_now + eps`.

Example of (a):
https://github.com/RobotLocomotion/drake/blob/4ec8e908309d1f6d23d813ccac2aa8fb1c0e3e14/systems/lcm/lcm_interface_system.cc#L59-L64

Example of (b)
https://github.com/RobotLocomotion/drake/blob/4ec8e908309d1f6d23d813ccac2aa8fb1c0e3e14/automotive/maliput_railcar.cc#L364-L369

(a)'s usage is well-documented with the current state of affairs.
(b), however, seems like a vestigial implementation.

Question for y'all @sherm1 @edrumwri @jwnimmer-tri:
Should (b) effectively be updated to be like (a), as it's effectively requesting an instantaneous timed event?

edrumwri, 2019-05-07:
Any instances where `eps` is added should be killed with prejudice.

That was necessary to work around a limitation in the old event system.

--------------------------------
Evan Drumwright
Senior Research Scientist
http://positronicslab.github.io
Toyota Research Institute
Palo Alto, CA

jwnimmer-tri, 2021-11-11:
Grepping through Drake master, all of the eps are gone now.

closed 2021-11-11 by jwnimmer-tri