RobotLocomotion/drake #11424: revert followup: Figure out why `custom_test` fails intermittently due to #11416

EricCousineau-TRI, 2019-05-09:
Relevant Slack convo: https://drakedevelopers.slack.com/archives/C270MN28G/p1557346698044300

I believe the repro recipe is using GCC:
```
bazel test -c dbg --runs_per_test=50 //bindings/pydrake/systems:py/custom_test
```

\cc @RussTedrake @jwnimmer-tri

EricCousineau-TRI, 2019-05-09:
Huh, tried reproducing it in our Xenial Docker build recipe @ cabd2ab, but to no avail - at least per the above command...

@jwnimmer-tri Will see if I can pick your brain tomorrow.

EricCousineau-TRI, 2019-05-16:
Per f2f with Jamie, this test failed again on CI on `master`.

If it helps at all, instructions for `valgrind`:
https://gist.github.com/EricCousineau-TRI/ce79d3265bb72934267e24ddc8c623bc

I build it, use `virtualenv` to point to that, install deps, and add `--python_path` and `--action_env=DRAKE_PYTHON_BIN_PATH` in `user.bazelrc`.

EricCousineau-TRI, 2019-05-23:
Per f2f, @BetsyMcPhail can repro on machine with Xenial installed directly. Should could not repro for running `custom_test` multiple times (even 1000x), but she found that running `test //...` for the above recipe *does* cause the `custom_test` to fail...

EricCousineau-TRI, 2019-06-06:
Still very difficult to reproduce, but we should be able to check the results of running it under `valgrind` to snag the memory leak, or whatever it is.

BetsyMcPhail, 2019-06-13:
I ran valgrind per Eric's instructions above - currently sorting through the output to determine which, if any, reported errors are real.

EricCousineau-TRI, 2019-06-13:
Sweet! Can you post a Gist of the output?

EricCousineau-TRI, 2019-06-13:
Per f2f, Betsy used the Python+NumPy suppressions; may try one more time to run with Drake suppressions before posting output.

BetsyMcPhail, 2019-06-14:
valgrind output for custom_test: https://gist.github.com/BetsyMcPhail/c2e223f65b7fff11c44f04e0253e21cb

EricCousineau-TRI, 2019-06-14:
Hm... That's quite a few errors, and but they all look like [just leaks](http://valgrind.org/docs/manual/mc-manual.html#mc-manual.leaks).

Just check, can you post the relevant reproduction recipe, including this stuff?
* The Drake source SHA, with mods included (e.g. if including other suppressions)
* Any mods to the Python+valgrind setup, if necessary
* The full command-line for your `bazel run` command-line

BetsyMcPhail, 2019-06-20:
1. Checkout `cabd2a`
  - Update `tools/dynamic_analysis/bazel.rc` so we can run memcheck on Python tests:
```
-build:memcheck --test_lang_filters=-sh,-py
+build:memcheck --test_lang_filters=-sh
```
 -  Update `tools/dynamic_analysis/valgrind.sh` to include any extra suppression files (e.g. python-debug/cpython/Misc/valgrind-python.supp and [customtest_valgrind.supp](https://gist.github.com/BetsyMcPhail/01f3eb693d6e698230d4f4edb99b9e52))

2. Follow Python+Valgrind setup, modifications:
   - Need a 'make' / 'make install' after the 'configure' in the first step
   - Skip GDB setup

3. ```bazel test --config=memcheck //bindings/pydrake/systems:py/custom_test```

EricCousineau-TRI, 2019-06-20:
Per f2f, trying to run all tests under DRD + Hellgrind.

jwnimmer-tri, 2019-06-27:
Just as my 2c: this doesn't smell like a memory or race error to me.  It smells more like a multiple-inheritance / ODR / python `imp` kind of error.  If I were debugging this, my next step would be to add more instrumentation to the test to figure out _where_ and _why_ the `assertTrue(called_publish)` is ending up false, and if anything else is broken, etc.

EricCousineau-TRI, 2019-06-27:
FTR Most likely relates workaround in here: https://github.com/RobotLocomotion/drake/pull/11719
Slack convo: https://drakedevelopers.slack.com/archives/C270MN28G/p1561403942037200

BetsyMcPhail, 2019-07-02:
What we know so far:

The ['test_all_systems_leaf_overrides' test](https://github.com/RobotLocomotion/drake/blob/967aba6524fc4bb8b338876754c7d48b25bcde89/bindings/pydrake/systems/test/custom_test.py#L224) is failing because the ['called_publish' flag isn't set.]( 
https://github.com/RobotLocomotion/drake/blob/967aba6524fc4bb8b338876754c7d48b25bcde89/bindings/pydrake/systems/test/custom_test.py#L360) The 'called_publish' flag should be set in the ['TrivialSystem::DoPublish'](https://github.com/RobotLocomotion/drake/blob/967aba6524fc4bb8b338876754c7d48b25bcde89/bindings/pydrake/systems/test/custom_test.py#L279) function that is is implemented as part of the test case.

To reproduce on Bionic:
* Check out 12a8432
* `CC=clang CXX=clang++ bazel test -c dbg --config=python3 //bindings/pydrake/systems:py/custom_test` Fail!
* In custom_test.py, rename'test_leaf_system_overrides' to 'test_all_leaf_system_overrides' 
* `CC=clang CXX=clang++ bazel test -c dbg --config=python3 //bindings/pydrake/systems:py/custom_test` Passes!

Following the code through, it's possible to see that in the [PYDRAKE_TRY_PROTECTED_OVERLOAD ](https://github.com/RobotLocomotion/drake/blob/967aba6524fc4bb8b338876754c7d48b25bcde89/bindings/pydrake/systems/framework_py_systems.cc#L79) macro, the first call to `PYBIND11_OVERLOAD_INT` macro doesn't return as expected.

```
#define PYDRAKE_TRY_PROTECTED_OVERLOAD(RETURN, CLASS, NAME, ...)       \
  PYBIND11_OVERLOAD_INT(RETURN, CLASS, NAME, __VA_ARGS__);             \
  if (py::get_overload<CLASS>(this, "_" NAME)) {                       \
    WarnDeprecated(DeprecatedProtectedAliasMessage(NAME, "override")); \
    PYBIND11_OVERLOAD_INT(RETURN, CLASS, "_" NAME, __VA_ARGS__);       \
  }
```

Other observations:
* If TrivialSystems::DoPublish is renamed to TrivialSystems::_DoPublish (note leading underscore), the test passes.
* If PYDRAKE_TRY_PROTECTED_OVERLOAD is replaced with PYBIND11_OVERLOAD_INT in [PyLeafSystemBase::DoPublish](https://github.com/RobotLocomotion/drake/blob/967aba6524fc4bb8b338876754c7d48b25bcde89/bindings/pydrake/systems/framework_py_systems.cc#L165), the test passes
* Possibly relates to #10445?
* Can't reproduce on current master?

BetsyMcPhail, 2019-07-18:
I found that for both `cabd2a` and `12a8432`, if 'DoPublish' is removed from [AddDeprecatedProtectedAliases](https://github.com/RobotLocomotion/drake/blob/3106f0599080422c40f0d74c36ff2854eb5d5659/bindings/pydrake/systems/framework_py_systems.cc#L753), the test behaves as expected. Per comments in the [source](https://github.com/RobotLocomotion/drake/blob/3106f0599080422c40f0d74c36ff2854eb5d5659/bindings/pydrake/systems/framework_py_systems.cc#L35), this function is set to be deprecated soon.

EricCousineau-TRI, 2019-08-01:
Lowering priority due to deprecation, but would still be nice to root-cause given reproduciblity at a given SHA.

EricCousineau-TRI, 2019-08-08:
Per f2f with Betsy and Bill, may be due to caching of overload lookup:
https://github.com/RobotLocomotion/pybind11/blob/4b8e231e7c209e5483b7ca2407a8212a30507277/include/pybind11/pybind11.h#L2399-L2454

BetsyMcPhail, 2019-08-13:
In the code snippet linked above, Pybind11 caches functions that aren't overloaded in Python to avoid costly Python dictionary lookups. The cache is a map from _(type.ptr, function name)_ to _function_. The cache is NOT cleared during execution of the tests.

Debug output from **test_deprecated_protected_aliases**. Note that 'DoPublish' is added to the 'inactive_overload_cache'  for 'OldSystem'

```
****************************** get_type_overload ***************************************
 *** self =  <custom_test.TestCustom.test_deprecated_protected_aliases.<locals>.OldSystem object at 0x7fe586057f48>
 *** self.ptr() = 0x7fe586057f48
 *** type =  <class 'custom_test.TestCustom.test_deprecated_protected_aliases.<locals>.OldSystem'>
 *** type.ptr() = 0x2746c58
 *** LOOKING FOR 0x2746c58 DoPublish
+++++++++++++++ CONTENTS OF CACHE ++++++++++++++++++++++
0x2719c38 _DoHasDirectFeedthrough
0x2719c38 DoHasDirectFeedthrough
+++++++++++++++++++++++++++++++++++++++++++++++++++++++
 *** overload is a c++ function, updating cache.
 
****************************** get_type_overload ***************************************
 *** self =  <custom_test.TestCustom.test_deprecated_protected_aliases.<locals>.OldSystem object at 0x7fe586057f48>
 *** self.ptr() = 0x7fe586057f48
 *** type =  <class 'custom_test.TestCustom.test_deprecated_protected_aliases.<locals>.OldSystem'>
 *** type.ptr() = 0x2746c58
 *** LOOKING FOR 0x2746c58 DoPublish
+++++++++++++++ CONTENTS OF CACHE ++++++++++++++++++++++
0x2746c58 DoPublish
0x2719c38 DoHasDirectFeedthrough
0x2719c38 _DoHasDirectFeedthrough
+++++++++++++++++++++++++++++++++++++++++++++++++++++++
 *** found key in non-overload cache.

```

Debug output from **test_all_leaf_system_overrides**.  Note we're checking for 'DoPublish' with the same key as above - even though it's a different class!!

```
 ****************************** get_type_overload ***************************************
 *** self =  <custom_test.TestCustom.test_leaf_system_overrides.<locals>.TrivialSystem object at 0x7fe5860ae048>
 *** self.ptr() = 0x7fe5860ae048
 *** type =  <class 'custom_test.TestCustom.test_leaf_system_overrides.<locals>.TrivialSystem'>
 *** type.ptr() = 0x2746c58
 *** LOOKING FOR 0x2746c58 DoPublish
+++++++++++++++ CONTENTS OF CACHE ++++++++++++++++++++++
0x2746c58 DoPublish
0x2719c38 DoHasDirectFeedthrough
0x2719c38 _DoHasDirectFeedthrough
+++++++++++++++++++++++++++++++++++++++++++++++++++++++
 *** found key in non-overload cache.
```

BetsyMcPhail, 2019-09-04:
High-level overview and proposed solution

Pybind11 caches functions that aren't overloaded in Python to avoid costly Python dictionary lookups. 

The non-overload cache is a set which uses _(type.ptr, function name)_ as a key. Once an item is added to the cache, it is never removed.

The bug occurs when a type that has been added to the cache (e.g type =  <class 'custom_test.TestCustom.test_deprecated_protected_aliases.<locals>.OldSystem'>
 type.ptr() = 0x2746c58) is destroyed and a new type is created with the exact same pointer (e.g. type =  <class 'custom_test.TestCustom.test_leaf_system_overrides.<locals>.TrivialSystem'>, type.ptr() = 0x2746c58) that should _not_ be in the cache. 

The first attempt at fixing the issue was to use _self.ptr_ instead of _type.ptr_ in the cache key. Testing demonstrated that this was not enough to solve the issue.

In the end, removing all associated cached items when an instance is deregistered seems to fix the issue.

EricCousineau-TRI, 2019-09-04:
Gotcha, nice! Another option would be to just extend the key `(type.ptr, type_full_name, function name)`, where `type_full_name` is something like `f"{cls.__module__}.{cls.__qualname__}".

EricCousineau-TRI, 2019-09-26:
Per #12105, seems like we have new a new issue with (potentially non-deterministic) segfaults.

PR #12095 (2019-09-25): pybind11: Use workaround for false cache hits for overloads
Resolves #11424

Requires:
- [x] Experimental CI to pass with PR
- [x] Merge of https://github.com/robotlocomotion/pybind11/pull/32

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12095)
PR #12589 (2020-01-23): pybind11: Prevent false cache hits for overloads
Resolves #11424 with RobotLocomotion/pybind11#37

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12589)
closed 2020-01-23 by ggould-tri