RobotLocomotion/drake #11448: need better error message for MBP users who (from their perspective) suddenly need SceneGraph

RussTedrake, 2019-05-13:
I'm seeing a lot of questions like this one below.  Clearly, the need for SceneGraph after working with MBPs is coming as a surprise to folks.  

> I also keep running into this error and I'm wondering what the correct way to resolve it is. I've taken some guesses, but none have worked:
```
RuntimeError: InputPort::Eval(): required InputPort[0] (geometry_query) of System ::drake/multibody/MultibodyPlant@0000000004171ff0 (MultibodyPlant<drake::AutoDiffXd>) is not connected
```

sherm1, 2020-06-15:
Relates to #10874

joemasterjohn, 2020-06-17:
#13558 adds a less vague error message:
>The geometry query input port (see MultibodyPlant::get_geometry_query_input_port()) of this MultibodyPlant is not connected. Please connect the geometry query output port of a SceneGraph object (see SceneGraph::get_query_output_port()) to this plants input port in a Diagram.

Does that suffice?

RussTedrake, 2020-06-23:
Thanks @joemasterjohn .  It definitely helps them with a resolution, but might there be a little preamble that helps them understand why the sudden requirement?  something as simple as "you've requested an MBP call that requires a geometry query... " then your text?

jwnimmer-tri, 2022-04-26:
See also https://github.com/RobotLocomotion/drake/issues/10874#issue-419049303 and https://github.com/RobotLocomotion/drake/issues/10874#issuecomment-471225736.  As part of this resolution, we should also consider updating our examples to use the sugar that always creates an MbP and SG as a single unit.

jwnimmer-tri, 2022-07-19:
After #17580 lands, what work or cleanup still remains open here?

SeanCurtis-TRI, 2022-07-19:
I believe all of the strategies discussed in this issue have been addressed by the various issues. #17580 didn't officially close this because I wasn't sure about merging order. I'm happy to update that PR so it automatically closes this.

SeanCurtis-TRI, 2022-07-19:
As the various solutions have been addressed in merged commits, we'll close this issue. If this crops up again, we can re-open it.

closed 2022-07-19 by SeanCurtis-TRI