RobotLocomotion/drake #11449: CalcDiscreteVariableUpdates needs better error message if same context is passed in twice

RussTedrake, 2019-05-13:
calling `system.CalcDiscreteVariableUpdates(context, context.get_mutable_discrete_state())` appears to simply take down the python interpreter -- presumably via a segfault.

Clearly, you should not pass the SAME context into both arguments, but for new users this is a confusing trap with no helpful diagnostics.

I assume this would be true for the time derivatives methods, too.

sherm1, 2019-05-13:
Is this the same as in #11450 where the user was working with the context of an embedded subsystem within a diagram? Although using the same context for both input and output here is not recommended, I don't think it should cause a fatal error. It would cause an immediate Context update and unpredictable ordering effects, but nothing fatal.

RussTedrake, 2019-05-13:
it was indeed the same user as #11450 (albeit at different times), so perhaps it was indeed the same root cause.  Do we consider the unpredictable ordering effects to be something we should protect the user from?  Or is this an acceptable pattern?

sherm1, 2019-05-13:
In the Simulator, ordering uncertainty would be a problem. But the Simulator will never use the same context for both arguments. The use case here is a direct call by a user to `CalcDiscreteVariableUpdates()` so I'm not sure there are any ordering issues to worry about. It's possible that the override is written to depend on disjoint memory for input and output in which case this would break. But if the user is invoking his own override here, I'm not sure we need to babysit. We don't currently say anything about the potential overlap of input and output argument memory, which is common for C++ methods (and can certainly cause trouble, but usually doesn't).

closed 2019-05-14 by RussTedrake