RobotLocomotion/drake #11450: plant.CalcDiscreteValueUpdates() crashes python (with no helpful error) if a SceneGraph is attached

RussTedrake, 2019-05-13:
Another new user pain point -- they call a method on the plant, passing in only the plant's context.  But the plant is a subsystem in a diagram, with (most critically) a SceneGraph attached.  I will try to reproduce the exact failure, but anecdotally, the user is confused because the python interpreter simply crashes with no helpful error.

e.g. 
```
    builder = DiagramBuilder()
    plant, scene_graph = AddMultibodyPlantSceneGraph(builder, MultibodyPlant(0.0005))
    # Build the plant
    parser = Parser(plant=plant, scene_graph=scene_graph)
    parser.AddModelFromFile("set_4/raibert_hopper_2d.sdf")
    plant.WeldFrames(plant.world_frame(), plant.GetFrameByName("ground"))
    plant.AddForceElement(UniformGravityFieldElement())
    plant.Finalize(scene_graph)

diagram = builder.Build()
    context = plant.CreateDefaultContext()
    plant.CalcDiscreteVariableUpdates(
            context = context,
            discrete_state = context_update.get_mutable_discrete_state())
```

EricCousineau-TRI, 2020-05-02:
FYI @sherm1 This is still reproducible (e.g. segfault), with Drake-only resources.
Used the same setup as mentioned here: https://github.com/RobotLocomotion/drake/issues/11457#issuecomment-622985019

Drake version:
```
In [1]: !cat /opt/drake/share/doc/drake/VERSION.TXT
Out [1]: 20200430074549 26c38c92d871567f333eb8ece2677bcfc4ab165f
```

sherm1, 2020-06-14:
This was fixed in #13536.

closed 2020-06-14 by sherm1