RobotLocomotion/drake #11457: DiagramContext.get_mutable_discrete_state failure

RussTedrake, 2019-05-14:
The following code 
```
from pydrake.all import (
    DiagramBuilder, 
    FindResourceOrThrow,
    Parser,
    MultibodyPlant,
    UniformGravityFieldElement,
)
from pydrake.geometry import DispatchLoadMessage, SceneGraph
from pydrake.multibody.plant import AddMultibodyPlantSceneGraph

if __name__ == '__main__':
    builder = DiagramBuilder()
    plant, scene_graph = AddMultibodyPlantSceneGraph(builder, MultibodyPlant(0.0005))
    # Build the plant
    parser = Parser(plant=plant, scene_graph=scene_graph)
    parser.AddModelFromFile(FindResourceOrThrow("drake/examples/acrobot/Acrobot.urdf"))
    plant.AddForceElement(UniformGravityFieldElement())
    plant.Finalize(scene_graph)

    # The diagram
    builder.ExportInput(plant.get_actuation_input_port())
    diagram = builder.Build()
    context = diagram.CreateDefaultContext()
    context.FixInputPort(0, [0])
    new_context = context.Clone()
    discrete_values = new_context.get_mutable_discrete_state()
#    discrete_values = diagram.AllocateDiscreteVariables()
    diagram.CalcDiscreteVariableUpdates(
            context = context,
            discrete_state = discrete_values)
```
fails with 
```
Exception has occurred: SystemExit
Failure at bazel-out/darwin-opt/bin/systems/framework/_virtual_includes/diagram/drake/systems/framework/diagram.h:1156 in DispatchDiscreteVariableUpdateHandler(): condition 'diagram_discrete' failed.
  File "/Users/russt/Downloads/test_hopper.py", line 32, in <module>
    discrete_state = discrete_values)
...
```

but if I pass in the result of `AllocateDiscreteVariables` instead (by uncommenting that line), then everything works as expected.  I don't understand why.  I dug into the call stack a bit, and it appears that the dynamic_cast to `DiagramDiscreteValues<T>` is failing on the first case.

EricCousineau-TRI, 2020-05-02:
@sherm1 FYI I am still able to reproduce this.

Easy repro steps:
- Go to Drake binder: https://mybinder.org/v2/gh/RobotLocomotion/drake/nightly-release?filepath=tutorials
- Create a new notebook: New (top-right) > Python 3
- Paste in code
- Run the cell (Shift+Enter)
- Once it fails, make a new cell, type in `%tb`, and press Shift+Enter.

Version info (from notebook):
```
In [3]: !cat /opt/drake/share/doc/drake/VERSION.TXT
Out [3]: 20200420074541 b02ff6e7b75d1a9a3ae7d4cf21fe76fa558cff74
```

sherm1, 2020-06-10:
I see now that neither the original `diagram.AllocateContext()` (`CreateDefaultContext()`)nor `diagram_context.Clone()` made any use of DiagramDiscreteValues -- both go through the same DiagramState code. Only `diagram.AllocateDiscreteVariables()` is doing this correctly.

PR #13536 (2020-06-12): Make DiagramContext use the DiagramDiscreteValues type
DiagramContext used the continuous variable object type DiagramContinuousState but failed to keep the discrete variables in a DiagramDiscreteValues object, instead flatting them into a plain DiscreteValues object. In contrast diagram.AllocateDiscreteVariables() does return a DiagramDiscreteValues object, of the kind required for doing discrete variable updates. The result is that you can't do an update directly into the discrete state of a diagram context, as would reasonably be expected.

This PR fixes that and adds missing unit tests tha [... cut]
closed 2020-06-12 by RussTedrake