RobotLocomotion/drake #11469: Should LeafSystem variables be hidden by default in Diagrams?

sherm1, 2019-05-15:
While reworking the variables per #9171 it occurred to me that declared variables in subsystems are often internal to those subsystems and are not generally meaningful at the Diagram level. For input and output ports we require explicit Export() calls to raise subsystem port visibility to the Diagram level, where they get their own names and port numbers.

We could do the same thing with parameters and state variables rather than simply collecting them together in bulk at the Diagram level. The bulk view is still required for computational purposes (for integration, e.g.) but we could prevent meaningless leakage of, say, energy integrals, the integrated error in PID controllers, and friction bristle states for hydroelastics. And we could flatten and provide nice names for important variables and parameters that might be buried several subsystem layers deep.

Any thoughts on this? It would be easy to do now and analogous to the port API, but I'm not sure it matters.

jwnimmer-tri, 2021-11-11:
I'm pretty sure that a fundamental assumption of our analysis toolbox is that when you ask a System (aka Diagram) for what kinds of state it has (and how much of that there is), that it gives you an honest answer.

How about we close this?

sherm1, 2021-11-11:
This proposal wouldn't preclude finding all the internal state for algorithms that require it (like the integrators mentioned above). It just wouldn't expose the internals by default. However, I think we can just rethink this if/when we rethink variables altogether per #9171. The idea is mentioned there and we don't need a separate issue here.

closed 2021-11-11 by sherm1

jwnimmer-tri, 2021-11-11:
(My key point would be that the analysis algorithm has no idea that the System it is interrogating is-a Diagram that would require extra work, and that we should not _need_ to teach it to know about that distinction and handle it.)