RobotLocomotion/drake #11470: beautiful ImplicitStribeckDocumentation is effectively undiscoverable

RussTedrake, 2019-05-15:
I get a lot of questions about drake's contact model / solver implementation.  I finally went looking for the doc.  I had to work really hard to find it!  And it's quite nice:
https://drake.mit.edu/doxygen_cxx/classdrake_1_1multibody_1_1_implicit_stribeck_solver.html

I don't think any novice could have found it.  Can we please make it more prominent?

SeanCurtis-TRI, 2019-05-15:
Please take a look at #11431 and see if that doesn't resolve the issue.

RussTedrake, 2019-05-15:
it does indeed.  hoorah!

amcastro-tri, 2019-05-15:
@RussTedrake, #11431 just merged. There is an entry point to contact modeling in the [multibody_plant.h here](https://github.com/RobotLocomotion/drake/blob/7b71e21463003f02aba31e2d9138ce8633fdec86/multibody/plant/multibody_plant.h#L164)

PR #11431 (2019-05-15): Contact modeling docs.
Resolves #11319.
Resolves #11470.

Brings back the awesome pieces of docs written by @SeanCurtis-TRI. 
New section on the physics of contact has been added and older sections were brought up-to-date with our current code. Removed no longer relevant sections.

Priority is high given the high demand for this from users.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11431)
closed 2019-05-15 by amcastro-tri