RobotLocomotion/drake #11476: Cache invalidation on Context cloning

SeanCurtis-TRI, 2019-05-15:
## Problem statement

Currently, when a `Context` is cloned, the "up-to-dateness" of the cache entries is likewise copied directly. The correctness of this approach is dependent on the *quality* of the copy semantics of all entries stored in cache entries (are all fields copied? Is all state updated appropriately?) Ultimately, Drake cache entries will accept any type that provides a copyable interface. But it can't guarantee the quality of that copy. Drake *does* have the demand that calling the `Calc()` method associated with the cache entry must guarantee the value's correctness.

## Proposal

When a `Context` is cloned, we'll set all of the cache entries to be invalid such that the next call to `Eval()` will require a `Calc()` call as well. This is a conservative approach that is consistent with the cache philosophy of "When in doubt, make sure the caching is not what makes the answer wrong."

## Points for discussion

Obviously, this works best when re-calc'ing the section of the cloned `Context` is a reasonable computational cost. There could be scenarios where the cost of calc'ing crossed with the frequency of cloning could kill performance for an application.

SeanCurtis-TRI, 2019-05-15:
cc @ggould-tri

ggould-tri, 2019-05-15:
This strikes me as a very good conservative approach.  I think I'm the only major consumer of context-cloning, and I do so only in computing initial conditions, ie before there's much of value in the cache anyway.  :+1:

sherm1, 2019-05-15:
I've got a fix in progress -- will push shortly and ask you to feature review, @ggould-tri, to make sure it fixes your troubles.

sherm1, 2019-05-15:
Posting for @jwnimmer-tri:

I am extremely disturbed by the hypothesis that user-supplied data in a Context does not suitably copy all values during cloning.

There may be a good reason why Cloning needs to invalidate a Context (e.g., if it doesn't copy an input port, values that depend on that input should be invalidated), but we should not even pretend support users doing `Value<Foo>` where `Foo(const Foo&)` (or `Foo::Clone`) is a lie.  There are plenty more places in the framework that we rely on value-cloning to work.

Or, if the problem is that a subclass of Context fails to copy true, then that's a broken DoClone.. on the subclass, and not something that the framework should paper over.

TL;DR: There might be a good reason to tune invalidations across Context.Clone, but Sean's post doesn't provide any.

Thanks,
Jeremy Nimmer
Toyota Research Institute

SeanCurtis-TRI, 2019-05-15:
> There are plenty more places in the framework that we rely on value-cloning to work.

That's the single most compelling point. I'm not familiar with all of them. I know allocation from model is one of them, but, as a requirement for the cloned result, it's not as demanding. In that case, it doesn't specifically require that values get copied so much as structure. Although, there are cases where values being successfully copied is relied on.

So, that suggests whenever an `AbstractValue` gets cloned, we should *always* be able to rely on it being a legitimate copy.

sherm1, 2019-05-15:
BTW per f2f with Sean, while it is clear that anything T in a `Value<T>`  should be able to copy itself correctly, it is not clear that copying a cache entry should make use of that capability. See #11482 for a discussion.

sherm1, 2019-05-15:
Another forward from @jwnimmer-tri:

> I disagree that we should conservatively paper over developer error.  If the developer plays stupid games, they should win stupid prizes.  We can work on better diagnostics perhaps, but I strongly believe that we should not silently paper over mistakes.

> For the moment, I think Grant should be able to work around this by manually invalidating the questionable Context, eh?

> I think the framework itself should not incorporate work-arounds that hide errors from users.  And especially it should not promise to do so in its API documentation (I haven't checked if the PR does this or not).

ggould-tri, 2019-05-16:
I strongly disagree with @jwnimmer-tri here.  "If the developer plays stupid games, they should win stupid prizes" is simply not apropos to me as the winner of said stupid prize:  A deep-magic Drake developer did something odd, so a user's IK programs fail?  Do we think that that user has a path out of this other than Drake expertise and direct access to the #dynamics gang?

My understanding here is:
* It is demonstrably true that there are non-cloneable objects in the world in general and the cache in particular;
* It is nearly impossible for anyone other than the original developer of such a situation to even find it, let alone fix it;
* We do not at present have a way of testing for or detecting this situation;
* Caches hide errors and reduce debuggability because by definition the contract of a well-designed cache is that it is indiscernable and its effects are nonlocal in time and execution context.  Therefore validity errors in a cache are an order of magnitude worse than validity errors elsewhere in code and require more stringent precautions.
* User code to invalidate the cache is ugly, hard to understand, and even harder to verify, again because of the definition of a cache.
* Therefore the cache needs protection against cloning.

I'm convincicble to the contrary: if we had a good way of testing for and detecting cloning violations, that would be a different story.  Perhaps some mechanism where a cloned cache asserts key- and value-equality with its original for every entry?  But short of that invalidating a cache is a simple precaution against an error that we demonstrably do not fully everywhere avoid.

A longer-term effort to find ways to enforce and test cloneability would also probably be in order.

SeanCurtis-TRI, 2019-05-16:
I find myself swayed on both sides.

It is deeply unfortunate that my `QueryObject` stupid games is causing @ggould-tri to win my stupid prize. (We'll open a check out line where you can redeem the prize for something different.)

As I read @jwnimmer-tri's comment, I interpret it as users of Drake that extend Drake. If they, after the fact, create a bad cache entry value, they should pay that price. They need the feedback that they have done something stupid. And by automatically invalidating, we are "papering over" that misuse of a Drake `AbstractValue`. In fact, we're paying a performance cost to paper over that mis-use. And that *does* sound wrong.

That is *not* the same as Drake core doing stupid things. We need Drake to not do stupid things internally. However, de-stupidifying `QueryObject` is non-trivial. (Well, there's a fairly direct way to make it work -- finish of the TODO'd design -- but that will lead to copying and destruction of big blobs of memory.)

As to some of @ggould-tri's points:
  1. "Non-cloneable objects in the ...cache".  It's worth noting that we're talking about a very special case. To live in a cache entry, an object must have a declared cloning mechanism (`Clone()` or copy constructor). If you're missing that, you can't be in the cache (or port) at all. So, we have an initial mechanism to preclude "non-cloneable objects". So, we're doing this work to allow people to *lie* about something being cloneable. `QueryObject` is a weird special case because, for now, it clones itself into an invalid state (the lie).
  2. "Nearly impossible for anyone other than the original developer...to even find it, let alone fix it." As @jwnimmer-tri suggests, we have more dependencies on cloneable `AbstractValue` than just cache entries. If an end developer introduces a struct with bad cloning semantics, by masking one possible avenue to reveal this badness, we're making debugging it even harder. Bad clone semantics should break everywhere where we clone.

Ultimately, I'm persuaded that this approach is not the right way forward.

Instead, I'm going to go ahead and finish the `QueryObject` so that it is legitimately copyable. That at least will make it work in the current code. There is evidence that copying the geometry state is an undesirable cost (which is why we fixed how unrestricted updates copy state). However, copying contexts is probably a less frequent operation than event handling so this particular cost may be negligible.

Ultimately, I'd vote for a mechanism by which a CacheEntry can declare it's behavior on cloning (i.e., `set_invalid_on_clone()` or `allocate_on_clone()` or some such thing. This way, the author of a system who knows something very specific about its behavior and its values can make that determination without anything else in a diagram or framework having to make accommodations for it.

ggould-tri, 2019-05-16:
I like the notion of object opting in to cache clone behaviour because it might allow us to declare that objects that claim cloneability in this context must also declare an equality operator.  That would in turn allow us to check that their clone method follows a clone equality law.  Does that seem reasonable?

(I will admit my general bias that all objects in a caching scheme should be immutable, cloneable, equality-testable, and generally referentially transparent in all of the ways; I know that's now how we roll, but I like the idea of opting such pure objects into the nicer features of caching)

SeanCurtis-TRI, 2019-05-16:
Just to play devil's advocate. Requiring the end user to provide an equality operator doesn't help. They can lie in that just as much as lying in the cloning. Or, more charitably, they can be equally wrong and still wind up with inexplicable bugs. I fear that requiring the user-defined test mechanisms would only provide the illusion of security.

ggould-tri, 2019-05-16:
Yes, implicit in my suggestion is an assumption that it's easier to write a correct equality operator than a correct clone.  That might not be valid.

sherm1, 2019-05-16:
Maybe I'm just a pushover, but I buy _everyone's_ arguments above! I think objects that claim to be copyable should copy correctly, _and_ I think the caching system should defend itself against copying problems since (as Grant points out above) the problems tend to be remote in time and space so daunting for an end user to debug.

The place to catch copying problems for an object are in that object's unit tests, not by obscure failures in consumers of those objects.

jwnimmer-tri, 2021-05-05:
I stumbled back here as part of an unrelated issue search.

Upon a quick re-read, one worthwhile tip not mentioned above is the [DisableCaching](https://drake.mit.edu/doxygen_cxx/classdrake_1_1systems_1_1_context_base.html#a834bb093a64a07cdee915b85f4fb54f4) method.  If a user has a failing program that starts working again after calling that method, then a caching bug is implicated.

closed 2024-11-01 by jwnimmer-tri