RobotLocomotion/drake #11487: Check multibody::Frame<T> / Body / etc and multbody::MultibodyPlant<T> are consistent.

siyuanfeng-tri, 2019-05-16:
Because of reasons, I end up having a few similar, but different mulitbody plants around. I have to pass frames / bodies and plants around for various computations. So it would be nice to have MBP to check if the inputs are relevant, i.e. the passed in frame is from the same plant. I think the functionality is there, https://github.com/RobotLocomotion/drake/blob/master/multibody/tree/multibody_tree_element.h#L127
it just needs to be put in all call sites.

siyuanfeng-tri, 2019-05-16:
cc @amcastro-tri @sherm1 On a similar note, it'd be nice to sanity check pairs of MBP + context. 
In general, is there any mechanism to check whether context is compatible with system? The default implementation seems too loose for me. I would favor context stores the system that made it, but this probably violates a lot of your principles.

sherm1, 2019-05-16:
There is currently a firm prohibition against a Context having pointers into a System, but there is still room for a lot more validity checking than we current do. And certainly an MBT Element can be validated against its MBT (elements have back pointers to their containing tree).

jwnimmer-tri, 2024-03-16:
Closing as a duplicate of #17734, which has more discussion.

closed 2024-03-16 by jwnimmer-tri