RobotLocomotion/drake #11497: DependencyTicket documentation is very hard to find...

RussTedrake, 2019-05-18:
Please add a link to doxygen for the dependency ticket to the doxygen for at least `DeclareVectorOutputPort`

Per #11496, I wanted to add the dependency ticket to the compass gait.  I didn't actually end up finding any documentation on it after spending a few minutes clicking through doxygen.   I ended up searching through examples to find what i needed.

RussTedrake, 2019-05-19:
and a related comment from a 6.832 exit survey
> I also ran into troubles with the feedthrough detection. My system had no direct feedthrough components, but still complained of algebraic loops. I used the deprecated DoHasDirectFeedback() to resolve this, but didn't totally figure out how I was *supposed* to do it.

jwnimmer-tri, 2019-05-20:
Duplicate of (the last checkbox item of) #11179.

closed 2019-05-20 by jwnimmer-tri