RobotLocomotion/drake #11503: A support for running include-what-you-use (locally and on CI)

jamiesnape, 2019-05-20:
Especially useful for ironing out a few `libstdc++` vs. `libc++` issues that are not handled by `cpplint`.

jwnimmer-tri, 2019-05-20:
Having full-tree IWYU is also fine, but in my opinion a better answer for standard library includes is that we should teach our cpplint fork about more types and their includes (C++14, etc.).  It should be a fairly simple change.

jamiesnape, 2019-05-20:
> Having full-tree IWYU is also fine, but in my opinion a better answer for standard library includes is that we should teach our cpplint fork about more types and their includes (C++14, etc.). It should be a fairly simple change.

I do not disagree.

jamiesnape, 2020-02-17:
CC @EricCousineau-TRI since you asked me a few weeks ago for the issue number regarding `std::array`.

jwnimmer-tri, 2023-01-27:
This is sufficiently far down on the priorities list that we're going to close it as "not planned".

closed 2023-01-27 by jwnimmer-tri