RobotLocomotion/drake #11507: geometry: Isolate LCM dependency

jwnimmer-tri, 2019-05-20:
The `//multibody/plant:multibody_plant_code` depends on `//geometry:geometry_visualization`, and `//geometry:geometry_visualization` depends on `//lcm` -- oops, now MBP depends on LCM.

There is only one tiny function in `geometry_visulization.h` needed by MBP.  We should carve out that function into some other file, either an existing one or a new one.

jwnimmer-tri, 2019-05-20:
I'll happily do the PR, but the only hard part is choosing the new location (and name?), which I'll defer to @SeanCurtis-TRI.  Choosing the name is the only hard part though, so it probably makes sense for you do to the PR anyway.  But if you'd like me to, just tell me the (file)names to use.

SeanCurtis-TRI, 2019-05-20:
Thanks. I'll do it first thing in the morning. As you say, it's trivial.

PR #11515 (2019-05-21): Move MakeDrakeVisualizerProperties
This utility function was causing other code that used it to end up dependent on LCM. The solution is to move it out of geometry_visualization.

The movement led to a name change (since it's now presented as a more general utility.)  So, this changes all the call sites with the new, generic name.

Closes #11507.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11515)
closed 2019-05-21 by jwnimmer-tri