RobotLocomotion/drake #11519: Compute ContactSurface<T> for tet mesh vs. half space.

amcastro-tri, 2019-05-21:
Relates to #9905.

This will be used in for the case of a soft body represented by a mesh of tests against a rigid plane with analytical representation.

The method's API would be something like: 
- `ContactSurface<T> CalcContactSurfaceSoftMeshVsRigidHalfSpace(const SurfaceMesh<T>& mesh, const RigidTransform<T>& X_HM)`.
with `X_HM` the pose of the mesh frame M in the halfspace frame H, where we'd assume the normal to the plane is in the `Hz` axis.

cc'ing @sherm1, @edrumwri

SeanCurtis-TRI, 2019-05-21:
Alternatively, it might make sense to *not* have a special API and just go the same way we go with `CalcSignedDistanceToPoint()` in which the implementation either filters non mesh-box pairs or throws if they are present (and then that implementation can continue expand the types of pairs it supports with time).

amcastro-tri, 2019-05-21:
Sorry for the confusion. This is not public API in either the query object or proximity engine but a little internal computational method. It'd get called from the narrow phase.

sherm1, 2019-05-21:
This is different than our f2f discussion, Alejandro, though it does sound more interesting. We can get away with a simpler analytic sphere pressure field (where the pressure field scalar is just the depth and the gradient is just the outward direction). No tet mesh needed in that case. But a rigid half-plane/soft tet-mesh is more interesting, assuming we can come up with a decent pressure field & gradient field over the tet mesh. My thought is that we could have the analytic case up and running faster.

amcastro-tri, 2019-05-22:
We had a Slack discussion yesterday. Some of the highlights of that discussion:
1) @sherm1 is concerned that tet vs analytic level set is too cumbersome to implement. 
2) It is true the level set on a sphere/plane are analytic. However the integration on the contact patch must be performed numerically using the quadrature pipeline.
3) The simple tet solution proposed here solves problem (2) and is continuous, the one property that makes the hydroelastic model work with our integrators.
4) We'll ponder other options for another day. I'll implement on Thursday.

jwnimmer-tri, 2023-05-23:
It is difficult for me to map this back to something that would make sense to other developers or users.

Is there a way to phrase this ticket in terms of a user-facing feature?  (Some kind of contact mode?)  Or if this is no longer relevant, please close.

closed 2024-11-02 by jwnimmer-tri