RobotLocomotion/drake #11521: Wire up internal representation of tet meshes to their specs.

amcastro-tri, 2019-05-21:
Relates to #9905.

In `SceneGraph` we register shape specifications. Later on a `ShapeReifier` creates the underlying in-memory representation of this object used for computational purposes. 
For instance, `ProximityEngine<T>::Impl` implements `ShapeReifier` to create FCL representations. 
We need similar hooks for the hydroelastic model.

Per f2f with @SeanCurtis-TRI, it'd seem not all the elements to accomplish this are in place. Therefore @SeanCurtis-TRI suggested that the shortest path forward to implement the underlying tet representations would entail having a static `never_destroyed` object within our first spiral specific methods such as the one outlined in #11519 for tet sphere vs rigid half space.

## Some Implementations Details. Broad phase.

We can make use of the already in place broad phase in `ProximityEngine`. There are several example queries that show us the pattern to follow in order to implement `ComputeContactSurfaces()` with the broadphse hooked up.

`ComputePointPairPenetration()`  is the first example of a query between geometry pairs. For the last argument to `collide()`  we pass our costume `SingleCollisionCallback` which takes care of the narrow phase. Notice that for `collide()` (in contrast to `.distance()`) the narrow phase only gets called for objects potentially in penetration only (that's what we want for hydroelastic).

Another example is `ComputeSignedDistancePairwiseClosestPoints()`, which passes a **templated** callback on `<T>`. That's what we want to support `AutoDiffXd` (and other scalars).

edrumwri, 2019-05-21:
I don't understand why there should be a tet representation? I don't see what you gain by having a tet representation vs. a tessellated surface when one of the bodies is rigid and the other soft.

sherm1, 2019-05-21:
I'm not clear why the emphasis on tets here? For hydroelastic contact we need a pressure field representation. For simple shapes that doesn't require a tet mesh since we can represent the pressure field analytically. Of course we will need a more general representation soon, but perhaps can delay that to later in order to have something running sooner?

amcastro-tri, 2019-05-22:
Even if we do not use test for "spiral 0", some sort of reification (the SG way of creating the underlying computational representation) must take place.
Notice that the solution proposed by @SeanCurtis-TRI shortcuts the reification process but at the same time provides the appropriate hooks to use the SG pipeline. This is needed regardless of what shortcut we take.

amcastro-tri, 2019-06-03:
Progress: #11562 under review.

SeanCurtis-TRI, 2020-05-13:
I'm closing this. We clearly have hydroelastic volume representations of a huge swathe of primitives (capsule is conspicuously absent). They are fully available for use. So, there it is.

closed 2020-05-13 by SeanCurtis-TRI