RobotLocomotion/drake #11523: Hydroelastic body on a rigid half space example.

amcastro-tri, 2019-05-21:
This is more like the "Spiral 0" for the general solution outlined in #9905.
This very minimalist example will consist of a soft (in the hydroelastic model sense) body sliding (or rolling) down an inclined plane.

It'll exercise, in addition to the obviously needed computation of forces:
 1. Wiring of `MultibodyPlant` with `SceneGraph`.
 2. Support of `AutoDiffXd` all the way through the contact computation.
 3. Broadphase through `ProximityEngine`.
 3. Setup through SDF/URDF files.

The bare minimum list of tasks to accomplish this are:
 1. Generate tet meshes for primitive shapes. Sphere for instance, #11520.
 2. Specialized contact surface computation for soft tet vs rigid analytical shape. Sphere vs. half space for instance, #11519.
 3. Wiring up `SceneGraph`, #11521.
 4. Compute traction forces, #11485.
 5. Create this one particular example.

cc'in @sherm1, @edrumwri.

sherm1, 2019-05-21:
:+1: Sounds like a great example -- should require lots of pieces to work. Under "bare minimum" I think you could replace the first item with:
1. Generate pressure fields for primitive shapes. Sphere for instance, #11520.

That is, the bare minimum is to have a pressure field. More generally we want a pressure field represented on a tet mesh (or possibly as a level set) but for simple shape an analytic formulation of the pressure field should be sufficient.

amcastro-tri, 2019-05-22:
Please refer to discussion in #11519 regarding the particular choice of contact patch estimation.
Notice each item above corresponds to a single PR. For item (2), we still need a single PR, regardless of the particular shortcut we take. 
In total "spiral 0" consists of the 5 PRs above.

amcastro-tri, 2019-06-10:
Progress:
 - Mesh generator #11520 merged.
 - Level set/tet mesh intersection #11558 and #11599 merged. 
 - Hook up to SG #11521 submitted, in progress.

PR #12100 (2019-09-27): Incorporate hydroelastic modeling into the bouncing ball demo.
Resolves #11523.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/12100)
closed 2019-09-27 by amcastro-tri