RobotLocomotion/drake #11531: Figure out why `py_reference` is needed when return value is `T*` for certain objects

EricCousineau-TRI, 2019-05-23:
From r3 of #11409, it's still unclear to me as to why `py_reference` is needed for a `T*` arg to ensure the `keep_alive<>()` is applied:
https://github.com/RobotLocomotion/drake/blob/60f25138a69b38322c28c16ec946f92baa4a3aac/bindings/pydrake/geometry_py.cc#L113

We should figure this out more concretely.

EricCousineau-TRI, 2019-06-27:
May be a duplicate of #11406, or may be different in that it uses the `py::keep_alive<>` property vs. imperative `py::keep_alive_impl` call.

m-chaturvedi, 2019-11-19:
> it's still unclear to me as to why py_reference is needed for a T* arg to ensure the keep_alive<>() is applied.

`py_reference` is not needed for `keep_alive` to be applied.  `keep_alive` is about controlling the order of calls to memory management functions (constructors and destructors) and `py_reference` / `take_ownership` are about whether those constructors or destructors would be called by Python's garbage collector (via pybind11).

In our case, if we want the returned object to nurse the `this` object, and don't care about Python to take ownership, we should be using `keep_live` + `reference`, which is the same as `reference_internal`.  But it seems to me we want the return value to nurse the `builder`, so I think it should be `keep_alive<0, 2>()`.  `0, 1` would make the return object to nurse the module.
https://github.com/RobotLocomotion/drake/compare/master...m-chaturvedi:issue_11531?expand=1

In order to understand it I had created a small document:
https://gist.github.com/m-chaturvedi/3e4a4aba15eca1b4c2ba9f36e848653c
(Using ASAN would be the right way in order to have defined behavior, I tried it by building cpython and pybind11 with the support, but pybind11 [seems to have unresolved issues with ASAN](https://github.com/pybind/pybind11/issues/1510)).

EricCousineau-TRI, 2019-11-21:
> [...] would make the return object to nurse the module.

I'm not entirely sure about this part. Will check your notes.

ToffeeAlbina-TRI, 2020-04-02:
@EricCousineau-TRI  to read documentation, determine if this issue can be closed.

EricCousineau-TRI, 2020-04-09:
> [...] so I think it should be `keep_alive<0, 2>()`.  `0, 1` would make the return object to nurse the module.

I still don't think that `<0, 1>` implies that the module is the nurse. The module is simply the containing scope. TTBOMK, `<0, 1>` normally implies `self` for *methods of classes*, but does not imply the scope owner in other cases.

Will see if I can confirm that.

EricCousineau-TRI, 2020-04-09:
Tracing it out:
[`type_caster_generic::cast`](https://github.com/RobotLocomotion/pybind11/blob/cac3edd971c70469aa710c13d59f9c29c7f4651f/include/pybind11/cast.h#L573-L577) - handles policies
[`rec->impl = ...`](https://github.com/RobotLocomotion/pybind11/blob/cac3edd971c70469aa710c13d59f9c29c7f4651f/include/pybind11/pybind11.h#L140-L169) - handles input arguments conversion and output argument conversion, depends on `call.parent`

EricCousineau-TRI, 2020-04-09:
w.r.t. @m-chaturvedi's code:
https://gist.github.com/m-chaturvedi/3e4a4aba15eca1b4c2ba9f36e848653c

Yup, as shown the docs:
https://pybind11.readthedocs.io/en/stable/advanced/functions.html#return-value-policies
For `return_value_policy::automatic`:
> Default policy. This policy falls back to the policy `return_value_policy::take_ownership` when the return value is a pointer.
as Mmanu mentioned. As shown in his code, `take_ownership` + `keep_alive` works, but creates a double-free situation.

Meta feedback:
It would have been easier to zero in had there been links directly to primary sources (documentation), and if there hadn't been speculation about modules being involved in the keep alive behavior.

FYI @jamiesnape

closed 2020-04-09 by EricCousineau-TRI

m-chaturvedi, 2020-04-10:
@EricCousineau-TRI 
Sorry about that ! But I think I had conveyed that link to the documentation to you, and also communicated the fact that I was wrong about the speculation, f2f.  My speculation was due to the fact that Pybind11 don't explicitly mention what they do for modules.  I had however forgotten to update the issue.

EricCousineau-TRI, 2020-04-10:
Sounds good, thank you for following up!