RobotLocomotion/drake #11550: Remove SceneGraphInspector::GetCollisionCandidates() sugar method

SeanCurtis-TRI, 2019-05-28:
## Problem statement

The `SceneGraphInspector::GetCollisionCandidates()` method is pure sugar. It wraps up a number of publicly available methods into one call. It is only used in one place:

https://github.com/RobotLocomotion/drake/blob/07afd0e86d7c23438fed35580747391254c9fca0/multibody/optimization/static_equilibrium_problem.cc#L34-L35

Given this singular usage, it would be better for this use case to wrap the public API with its own sugar.

## Solution

1. Remove the `SceneGraphInspector` method (and directly supporting implementations).
2. Replace the call in the static equilibrium method with a locally-defined method.

avalenzu, 2019-05-29:
It's also called here:  https://github.com/RobotLocomotion/drake/blob/f7d755ad805cc7a732664ada3b91ac925251a140/multibody/inverse_kinematics/minimum_distance_constraint.cc#L50-L54

SeanCurtis-TRI, 2019-06-05:
Given that there are two functional users of this sugar, it probably makes sense to keep it. 

Feel free to re-open the issue if you disagree.

closed 2019-06-05 by SeanCurtis-TRI