RobotLocomotion/drake #11571: Use a faster BLAS/LAPACK library than the system one

hongkai-dai, 2019-06-04:
Many third-party libraries need BLAS/LAPACK, for example SCS /CSDP. Currently in Drake it builds with the system BLAS/LAPACK. I found in several cases that openblas in about 10x faster than the system BLAS, and Intel's MKL is even faster (20 to 100x faster than system blas). Can we change the drake build to use a different BLAS? On ubuntu we only need to do `sudo apt-get install libopenblas-dev` to install openblas.

jwnimmer-tri, 2019-06-04:
\CC @jamiesnape and @EricCousineau-TRI should be involved in this discussion.

EricCousineau-TRI, 2019-06-04:
Would prefer that we use system-installable versions if possible. Not super familiar with what MKL licensing might impose; mayhaps nothing?

That being said, before we do too much to Drake, I think I agree with @jwnimmer-tri (via f2f) that we should ideally test this on Anzu first.

hongkai-dai, 2019-06-04:
Thanks Eric.

Testing it in Anzu sounds a good plan. I think openblas is very easy to install on Ubuntu, via apt-get. Can we test openblas first? MKL is faster than openblas, but I imagine that to be more complicated (we need to AMD BLIS instead of Intel MKL if we detect AMD CPU).

jamiesnape, 2019-06-04:
Note Mac is already using OpenBLAS for any dependencies that need BLAS.

jwnimmer-tri, 2019-06-04:
Because Drake on Ubuntu uses pkg-config for `blas`, and because Ubuntu ships `blas.pc` via the `/etc/alternatives` mechanism, I think doing `sudo apt install libopenblas-dev` is all we need for openblas to end up being used by default.

jamiesnape, 2019-06-04:
Also https://software.intel.com/en-us/articles/installing-intel-free-libs-and-python-apt-repo.

jamiesnape, 2019-06-04:
> Because Drake on Ubuntu uses pkg-config for blas, and because Ubuntu ships blas.pc via the /etc/alternatives mechanism, I think doing sudo apt install libopenblas-dev is all we need for openblas to end up being used by default.

Exactly. It is a trivial change on Linux.

jwnimmer-tri, 2019-06-05:
@hongkai-dai Beware that when I (accidentally) switched my local Anzu to `libopenblas`, many regression tests fail because SNOPT produces different results (hitting an iteration limit) when the initial IIWA position is at (or near) zero for some or all joints.  This is a known hazard of IIWA initial positions for IK, but it's annoying that a BLAS upgrade tickles it.

hongkai-dai, 2019-06-05:
@jwnimmer-tri , thanks for the test. Is it possible to keep both system BLAS and openblas, and choose to build snopt with the system blas, and CSDP with openblas?

jwnimmer-tri, 2019-06-05:
Mixing different blas's would probably lead to insanity.

hongkai-dai, 2019-06-05:
I just did some test on my machine on some different IK problems. Even when I don't use the singular configuration as the initial seed, using different BLAS causes very different results. Sometimes SNOPT fails to solve with one BLAS but succeeds with the other.

jamiesnape, 2019-06-05:
> Mixing different blas's would probably lead to insanity.

We already have an example of that. It is a non-starter in general.

jamiesnape, 2019-06-06:
Are there any tests in Drake itself that we would expect to fail under OpenBLAS? Running with `--config=snopt --test_tag_filters=-gurobi,-mosek ...` on Mac is looking clean on my OpenBLAS branch.

EricCousineau-TRI, 2019-06-06:
Just curious: When you mentioned the cases where it ran 20-100x, can I ask which cases those were? Were they non-trivial problems?

jwnimmer-tri, 2019-06-06:
One example from the email thread related to this issue was a solve going from 100 seconds to 5 seconds.  So yes, non-trivial problems.

My (unexplored) hypothesis on the speedup was that the new library was eating all 40+ cores by default.

jamiesnape, 2019-06-06:
> My (unexplored) hypothesis on the speedup was that the new library was eating all 40+ cores by default.

If that is MKL, I would definitely think so.

hongkai-dai, 2019-06-06:
The test problem solves a big semidefinite programming problem. The speed bottleneck for this problem is to do Cholesky decomposition on a big matrix (2000 x 2000), which calls BLAS cholesky decomposition function.

When I link system blas, it also links to the system openmp. When I use MKL, I link to libgomp (GNU openmp).

hongkai-dai, 2019-06-06:
The 20x speed up was observed on my home desktop, which has only 4 cores.

jamiesnape, 2019-06-06:
(Note, at least on Mac, it appears MOSEK uses MKL internally.)

jwnimmer-tri, 2019-06-08:
I think the action item here is on @hongkai-dai to experiment to find a version he likes, and then we will prototype it inside Anzu.

EricCousineau-TRI, 2019-06-13:
(Switching team labels until action items come back to Kitware)

amcastro-tri, 2019-07-25:
@hongkai-dai, you said you tested openblas and MKL. Could you provide instructions on how to install and to link these libraries to Drake/Anzu? I'd like to do some testing myself for our contact solvers
Is there an easy way to switch back and forth between libraries? or would I need to uninstall/install every time I want to test a new library? (sry for my ignorance guys)

hongkai-dai, 2019-07-25:
@amcastro-tri I installed openblas through `sudo apt-get install libopenblas-dev`, and installed MKL by following their website https://software.intel.com/en-us/mkl/choose-download. I used these BLAS with CSDP solver. In Drake's build file, it just chooses the BLAS/LAPACK through pkg-config to find libblas.so https://github.com/RobotLocomotion/drake/blob/master/tools/workspace/libblas/repository.bzl. I haven't figured out how to use openblas for some library, and the Ubuntu system default blas for the rest of libraries yet. So if you install openblas/atlas/MKL, then EVERY code in drake (whose BUILD.bazel has `deps = @blas`) will switch to that BLAS. Some of the tests will fail when the code uses a different BLAS.

I just forwarded you the email that Jeremy and I discussed using different BLAS in Drake/Anzu

jwnimmer-tri, 2021-11-11:
I don't think there is any further action to take here.  On Ubuntu especially, I think it's easy for users to install a different blas and it will be used automatically.

closed 2021-11-11 by jwnimmer-tri