RobotLocomotion/drake #11573: parsers, sdformat: Enable custom (namespaced) Drake tags

EricCousineau-TRI, 2019-06-04:
(Did not seem like we have a tracking issue for this.)

We should ensure that we can parse custom tags (and specify a convention for adding custom tags).

Steps:
* Try out parsing a custom tag from `sdformat`, just to ensure that we still have it. (There were some fixes in the past, just need to double-check.)
* Define conventions. This should dovetail with a proposal to be submitted to `sdformat.org`.

\cc @sammy-tri @amcastro-tri 
FYI @scpeters @azeey

amcastro-tri, 2019-06-04:
Thanks @EricCousineau-TRI!. In particular we'll pretty soon need custom tags for material properties such as modulus of elasticity (aka Young's modulus).

EricCousineau-TRI, 2019-09-19:
@azeey has merged in a PR to the SDFormat spec to propose the parsing of custom tags. Once that is coded up and used in Drake, I can test it out, possibly on a toy example, and then close this out:
http://sdformat.org/tutorials?tut=custom_elements_attributes_proposal&cat=pose_semantics_docs&

amcastro-tri, 2019-09-20:
Question, that document does not describe how the `schema` is defined/written. Do you have a link to that?

When defining a custom entry, we'd ideally like to specify whether it is required (must be provided by a user or parser will throw) or if it has a default value (if not provided, then it assumes a default value) and what that value might be. Can we do that with the schema? how would the sdformat API support this?

jwnimmer-tri, 2019-09-20:
By definition, third-party custom tags in sdf files can never be required -- they are third-party!  A `foo.sdf` file found in the wild will never have those tags.

However, if the `drake::multibody::Parser` wants to write its own application-layer validation code beyond the XML schema, so that it _requires_ extra tags and will fail to load the thousands of SDF or URDF files already in existence, I guess that's your call.

amcastro-tri, 2019-09-20:
But it won't be in the wild right? our files most likely will define the namespace `xmlns:drake=//path/to/drake_custom_chema`, no?

jwnimmer-tri, 2019-09-20:
I think we're talking past each other.  I think the big picture answer is this: the libsdformat will give you the information from the XML file, unmodified.  If multibody needs defaults, error handling, or etc. beyond that then write C++ code in multibody to do that.  The sdformat library will not do those things.

jwnimmer-tri, 2019-10-30:
@EricCousineau-TRI I believe that in our parser, we are already loading custom tags (\CC @SeanCurtis-TRI).  What, if anything, remains to do on this issue?