RobotLocomotion/drake #11578: SortedPair<GeometryId> is not hashable

hongkai-dai, 2019-06-04:
I tried to use `SortedPair<GeometryId>` as a key for an unordered map as
```C++
std::unordered_map<SortedPair<GeometryId>, int> my_map;
SortedPair<GeometryId> pair;
my_map.emplace(pair, 0);
```
The compiler says `SortedPair<GeometryId>` is not hashable.

On the other hand, `GeometryId` is hashable, as `unordered_map<GeometryId, int>` works, but when we use `SortedPair<GeometryId>`, it is not hashable.

jwnimmer-tri, 2019-06-04:
You need to specify `drake::DefaultHash` for the `Hash` template argument of the `std::unordered_map`.

edrumwri, 2019-06-05:
I also had to add this to identifier.h to get it to work:

```
   /// Implements the @ref hash_append concept.
   template <class HashAlgorithm>
   friend void hash_append(HashAlgorithm& hasher, const Identifier& p) noexcept {
      using drake::hash_append;
     hash_append(hasher, p.get_value());
   }

```

jwnimmer-tri, 2019-06-05:
Ah yes, I see now that `geometry/identifier.h` does not follow Drake's rules as prescribed by https://drake.mit.edu/styleguide/cppguide.html#std_hash.  Perhaps it predated that rule.

PR #11872 (2019-08-01): Updates geometry::Identifier hash method
`geometry::Identifier` predated the drake hash protocol. It was made to hash but its hash mechanism apparently precluded it being used in other entities that *did* use the drake hash protocol.

This changes the hash mechanism to conform and explicitly tests that `SortedPair<Identifier<Tag>>` is also hashable (as called out in issue 11578).

resolves #11578

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11872)
closed 2019-08-01 by SeanCurtis-TRI