RobotLocomotion/drake #11580: [geometry] Define domain of geometries for ComputeSignedDistancePairwiseClosestPoints

SeanCurtis-TRI, 2019-06-05:
## Problem statement

`QueryObject::ComputeSignedDistancePairwiseClosestPoints()` computes the signed distance between all *unfiltered* geometry pairs. (There is an optional parameter that can exclude results if their distance is greater than a user-specified threshold.) Frequently, an application may only want signed distance for an enumerable set of pairs with particular interesting properties. The only way to achieve this currently is to compute all pairs and then throw out the results that aren't part of the enumerable set. Historically, this is how we originally achieved the "max distance" threshold as well -- however inclusion of the parameter has anecdotally realized the hoped-for computational performance). We would expect a similar performance improvement when the size of the target set is much smaller than the number of all unfiltered pairs.

## Proposal

We provide the following overload:

```c++
std::vector<SignedDistancePair> ComputeSignedDistancePairwiseClosestPoints(
    const std::set<SortedPair<GeometryId>>& pairs) const;
```
It has the following semantics:
  1. It respects collision filtering; if `pairs` contains pair `(A, B)` and `(A, B)` has been filtered then the results will _not_ include distance between A and B.
  2. The *order* of the output is not correlated with any implication of ordering on the input. Each returned result contains the `GeometryId`s of the geometries and this can easily be mapped back to the inputs.
  3. For an input `SortedPair<GeometryId> pair` that represents an *unfiltered* pair, we'll get back a `SignedDistancePair result`. We guarantee that `pair.first = result.id_A` and `pair.second = result.id_B`.

SeanCurtis-TRI, 2019-06-05:
cc @hongkai-dai

SeanCurtis-TRI, 2019-06-05:
## Additional thoughts

Related to the discussion in #10990, we could add a high layer of abstraction based on the concept of the `GeometrySet`.  We would have the following overloads:

```c++
/** Computes the O(N^2) signed distance measures between the N geometries contained in
`set`. (Minus those pairs that are marked filtered).  */
std::vector<SignedDistancePair> ComputeSignedDistancePairwiseClosestPoints(
    const GeometrySet& set) const;
```

```c++
/** Computes the O(MN) signed distance measures between the M and N geometries
 contained in `setA` and `setB`, respectively. (Minus those pairs that are marked filtered).  */
std::vector<SignedDistancePair> ComputeSignedDistancePairwiseClosestPoints(
    const GeometrySet& setA, const GeometrySet& setB) const;
```

That could lead to invocations like:
```c++
/// Compute distance  for pairs (A, B), (A, C), (B, C).
Compute...ClosestPoints({A, B, C});

/// Compute distance for pair (A, B).
Compute...ClosestPoints({A}, {B});

/// Compute distances for pairs (A, C), (A, D), (B, C), and (B, D)
Compute...ClosestPoints({A, B}, {C, D});

// Compute distances for all pairs (except for A)
GeometrySet setA{A};
Compute...ClosestPoints(setA.Not());

// Compute distance for all pairs of geometries that are both in two sets of
// interest.
GeometrySet set1{...};
GeometrySet set2{...};
Compute...ClosestPoints(set1.Intersection(set2));
```

*This* issue is about the initial proposal. If this follow up sugar has value, we'll spin it into its own issue.

sherm1, 2019-06-05:
Cool!

avalenzu, 2019-06-10:
Is the omission of a threshold distance in the proposed API intentional? If so, I'd be interested to hear the reasoning behind it.

SeanCurtis-TRI, 2019-06-10:
It was intentional. There was no strong agenda, but really a lack of imagination.

This provides one way of slicing the possible results. A maximum distance is another way to slice. I couldn't think of a use case where you'd slice both ways.

The need would be phrased something like:

> I know I will never care about any distances except those between: (A, B), (A, C), and (C, D) (even though there might be dozens of other geometries). However, in in the case of the geometry pairs that I care about, I only care about them if they are closer than 1 meter apart.

Is that the use case you have in mind? If so, adding the threshold would be reasonable.

avalenzu, 2019-06-10:
Yeah, that's the need I was thinking of, but I don't have a use case that exposes that need at present.

SeanCurtis-TRI, 2019-06-13:
Easy enough to include it from the get go.

dmsj, 2020-10-01:
To help with this API discussion, [here's an example](https://github.com/DexaiRobotics/drake-torch/blob/master/examples/drake-ompl/src/drake_ompl_demo.cc) of how we have used this feature for collision checking. 

We use a [lambda function](https://github.com/DexaiRobotics/drake-torch/blob/0130126fdb9e067fba9c61677de6def7d72ff0ee/examples/drake-ompl/src/drake_ompl_demo.cc#L209-L249) to return whether or not there is a collision. 

Is this the anticipated or suggested manner to use this API? It's not terribly verbose and relatively easy to use once you get a handle on the syntax.

We have also written a bunch of code to address this problem as stated by @SeanCurtis-TRI 

> I know I will never care about any distances except those between: (A, B), (A, C), and (C, D) (even though there might be dozens of other geometries). However, in in the case of the geometry pairs that I care about, I only care about them if they are closer than 1 meter apart.

For instance in the above example, we might want have different collision tolerances between various links of the robot and geometry in the environment.

SeanCurtis-TRI, 2020-10-01:
@dmsj Help me understand what I'm seeing in your link.

In the example, I see [this code](https://github.com/DexaiRobotics/drake-torch/blob/0130126fdb9e067fba9c61677de6def7d72ff0ee/examples/drake-ompl/src/drake_ompl_demo.cc#L209-L249) -- a lambda function that reports if any unfiltered geometry pair is closer than a threshold.

Questions:

1. Your still operating on the full world -- not trying to limit the geometries that are in consideration. This issue is about reducing the domain of *geometries* in consideration.
2. Why are you calling `ComputeSignedDistancePairwiseClosestPoints` with a distance tolerance of `FLAGS_max_distance` but then testing against a collision tolerance of `FLAGS_collision_tolerance`. You should just use the smaller of the two values and if you get *any* results, then you are in collision.
3. It seems like you could benefit from a method *similar* to `QueryObject::HasCollisions()` but takes a threshold. Something like `QueryObject::HasGeometriesCloserThan(double distance)`. Because, right now, you're computing *all* the distances even if the first two geometries you consider have that property.
4. As to you last point, "we might want have different collision tolerances between various links of the robot and geometry in the environment", I find that an intriguing proposition. Rather than a single global distance threshold, you have a per-pair distance threshold. Is that a concrete desire or a passing thought of, "Wouldn't that be a neat feature?" I can imagine an API that would supplant the `SortedPair` in the issue description with a `DistanceCandidate` class which includes: the two geometries and the minimum distance for them to be considered.  Similarly, a set/vector of those could be passed to `HasGeometriesCloserThan`. Just thinking aloud here....

dmsj, 2020-10-02:
Sean - sorry that was a bit of a rushed comment - I appreciated the thinking aloud! 

Answers to your points above:
1. Correct - however this latest issue I was tracking on this and related to my more complex desire of setting per object pair collision tolerances

2. Good point! This some confusion on my part combined with copying code into an example, when in practice we do the more complicated thing, which is check for all distances, and filter based on specific, per object collision tolerances. For example, some robots may collide with themselves, but their geometry and pose is well know, so we can use a very tight collision tolerance. Conversely, when testing for collision against a measured object in the environment, we have to account for uncertainty in the measured pose, so we use a larger tolerance. 

4. re: "wouldn't that be a nice feature?" Yes! It is a nice feature =) We have implemented this and it is a pretty important part of our stack (for the reasons mentioned in 3). We've basically implemented what you've proposed, in that we get all the signed distance pairs, then sort based on names, filter appropriately, and report a collision. 

On a related note, this also relates to collision filter groups, and we do define those on the fly, depending on the use case. For instance, we'll do things like:
```c++
GeometrySet g_world = robot_plant_->CollectRegisteredGeometries(
      robot_plant_->GetBodiesWeldedTo(robot_plant_->world_body()));
  scene_graph_->ExcludeCollisionsWithin(g_world);
```
To create our own filter groups as appropriate for the current set of geometry at hand. This API works fine, but we would like to get a list of existing collision filter groups and modify them, instead of not being able to easily query which collisions are already excluded. Again, thinking aloud, but this is very similar to your proposed `DistanceCandidate` class but based on exclusion as opposed to inclusion.