RobotLocomotion/drake #11581: Make DifferentialIK leaf system available within pydrake

naveenoid, 2019-06-05:
for examples/manipulation_station/differential_ik.py
Either implement in C++ and as the TODO within this file states or find some other way to expose this functionality to downstream consumers. Assigning to @RussTedrake since his names on the TODO.

EricCousineau-TRI, 2020-05-02:
FTR We have the function here:
https://drake.mit.edu/pydrake/pydrake.manipulation.planner.html#pydrake.manipulation.planner.DoDifferentialInverseKinematics

RussTedrake, 2020-07-17:
FTR -- goal is to move that system into c++ (and add the bindings)

jwnimmer-tri, 2022-06-27:
The file `drake/examples/manipulation_station/differential_ik.py` still exists (including its "puppetmaster" copy-pasta in Anzu, which cites this issue).

I am not totally sure from the above, but my impression was that prior to closing this ticket, we were supposed to delete that manipulation_station file and change the teleop tools to use `DifferentialInverseKinematicsIntegrator` instead.

(Re-opening so we don't lose track of this question.)

PR #13877 (2020-08-18): Adds DifferentialInverseKinematicsIntegrator
Cleaner version of the python LeafSystem DifferentialIK from the manipulation_station folder, now in c++.

Resolves #11581

Will send a follow-up PR to convert the manipulation station to this version.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/13877)
commit 5a9cd159 (2020-08-18): Adds DifferentialInverseKinematicsIntegrator (#13877)

Cleaner version of the python LeafSystem DifferentialIK from the manipulation_station folder, now in c++.

Resolves #11581

Will send a follow-up PR to convert the manipulation station to this version.
PR #17899 (2022-10-05): [examples/manipulation_station] Use DifferentialInverseKinematicsIntegrator
Removes the local implementation of a differential IK in favor of the proper Drake version.

Resolves #11581 (again).

+@naveenoid for feature review, please.  (Please make sure you can run the three notebooks)

- - -
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/17899)
closed 2022-10-05 by RussTedrake