RobotLocomotion/drake #11608: Rewrite MosekTest.SolverOptionsTest to not use MSK_DPAR_SEMIDEFINITE_TOL_APPROX

jamiesnape, 2019-06-11:
`MSK_DPAR_SEMIDEFINITE_TOL_APPROX` does not appear to behave as expected in MOSEK 9.0.

Blocks #11595.

hongkai-dai, 2019-06-11:
The IT department at TRI mentions that they will add the Mosek 9 license this month. I will rewrite that test once the TRI Mosek license is ready.

tri-ltyyu, 2019-06-20:
Got an update from Pete at IE. Working on it now and should have an update by tomorrow.

EricCousineau-TRI, 2019-06-27:
Is this one now potentially unblocked by the license upgrade? (checking convos, seems like we've got the upgrade?)

hongkai-dai, 2019-06-27:
@EricCousineau-TRI I was working on this last night, will send a patch today.

hongkai-dai, 2019-06-27:
@jamiesnape I cannot send a PR to your branch, so here is a patch for your PR #11595
[diff.txt](https://github.com/RobotLocomotion/drake/files/3335691/diff.txt), could you apply this patch on top of your PR?

jamiesnape, 2019-06-27:
Sure, thanks.

PR #11595 (2019-06-28): Upgrade to MOSEK 9.0.96
Unfortunately we cannot get good stack traces for the Valgrind suppressions in this version.

Fixes #11608.

----
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11595)
closed 2019-06-28 by jamiesnape