RobotLocomotion/drake #11610: C++17 support in CI

jwnimmer-tri, 2019-06-11:
We'd like to have Jenkins automatically test the Drake build in C++17 mode (instead of C++14), to find any future warnings or compatibility issues.  Some of our users build in C++17 mode.

I think for starters four new builds `linux-bionic-{clang,gcc}-cxx17-bazel-{continuous,experimental}-release` or so would be good enough for my purposes.

\CC @EricCousineau-TRI

jamiesnape, 2019-06-11:
Which compiler versions (same as currently or newer)?. Do you want us to add a config in Drake to switch to C++17 mode or do you want the `cxxopts` added on the command line on Drake CI for now?

jamiesnape, 2019-06-11:
Experimental jobs are here https://drake-jenkins.csail.mit.edu/view/C++17/. I have to make some changes to the configuration scripts before I can create continuous/nightly and/or weekly jobs so that build cops do not get deluged with "build failed" emails.

EricCousineau-TRI, 2019-06-13:
Once #11622 lands, Jamie can spin up the jobs.

Per f2f, Jamie could possibly switch C++17 on by default for Mac (if High Sierra does not block this).

jamiesnape, 2019-06-13:
Looks like High Sierra is going to be unhappy with C++17 (at least from an SDK point of view).

jwnimmer-tri, 2019-06-13:
(You probably know this already but) I think it's OK to omit support for C++17 on High Sierra.  We won't support C++17 on Ubuntu 16.04 with GCC 5, either.  Both platforms should evaporate in the next 5-6 months, anyway, I think.

jamiesnape, 2019-06-14:
We have experimental/nightly for Bionic Clang/GCC Bazel/CMake everything/default and Mojave Clang Bazel/CMake default. I will switch on some continuous on Monday.

EricCousineau-TRI, 2019-06-20:
Jamie will push the button, should be done today.

jamiesnape, 2019-06-20:
Continuous build have been created and are either in-progress or have completed successfully.

closed 2019-06-20 by jamiesnape