RobotLocomotion/drake #11620: dReal / ibex build errors in C++17 mode

jwnimmer-tri, 2019-06-12:
See logs in, e.g., https://drake-jenkins.csail.mit.edu/view/C++17/job/linux-bionic-gcc-bazel-experimental-cxx17-release/2/:
```console
[2019-06-11T20:26:38.840Z] In file included from external/dreal/opt/libibex/2.7.4/include/ibex/3rd/fp_traits/fp_traits_double.hpp:1384:0,
[2019-06-11T20:26:38.840Z]                  from external/dreal/opt/libibex/2.7.4/include/ibex/3rd/fp_traits/fp_traits.hpp:306,
[2019-06-11T20:26:38.840Z]                  from external/dreal/opt/libibex/2.7.4/include/ibex/3rd/interval/interval.hpp:38,
[2019-06-11T20:26:38.840Z]                  from external/dreal/opt/libibex/2.7.4/include/ibex/ibex_Interval.h:25,
[2019-06-11T20:26:38.840Z]                  from external/dreal/opt/libibex/2.7.4/include/ibex/ibex_Affine.h:15,
[2019-06-11T20:26:38.840Z]                  from external/dreal/opt/libibex/2.7.4/include/./ibex.h:7,
[2019-06-11T20:26:38.840Z]                  from external/dreal/opt/dreal/4.19.02.1/include/dreal/util/box.h:9,
[2019-06-11T20:26:38.840Z]                  from external/dreal/opt/dreal/4.19.02.1/include/dreal/api/api.h:7,
[2019-06-11T20:26:38.840Z]                  from external/dreal/opt/dreal/4.19.02.1/include/dreal/dreal.h:5,
[2019-06-11T20:26:38.840Z]                  from bazel-out/k8-opt/bin/solvers/_virtual_includes/dreal_solver/drake/solvers/dreal_solver.h:6,
[2019-06-11T20:26:38.840Z]                  from solvers/dreal_solver_common.cc:2:
[2019-06-11T20:26:38.840Z] external/dreal/opt/libibex/2.7.4/include/ibex/3rd/ieee/primitive.hpp:662:73: error: ISO C++1z does not allow dynamic exception specifications
[2019-06-11T20:26:38.840Z]   void readBitSet(std::istream & in, unsigned int n0, unsigned char * a) throw(interval_io_exception);
```

I don't think there is a flag to turn this off, and dReal is part of the default build, so it's a pretty serious disruption for any C++17 users who aren't using Bazel (where they could avoid the dReal dependency).

Can we build a forked / patched version of IBEX?

soonho-tri, 2019-06-12:
It's from an interval library that ibex is configured to use. I'll fix the issue and distribute a new version.

soonho-tri, 2019-06-13:
Fixed by #11635 .

closed 2019-06-13 by soonho-tri