RobotLocomotion/drake #11628: [geometry] Repackage classes in the MeshField family.

DamrongGuoy, 2019-06-12:
We would like to package the MeshField family in `geometry/proximity/` into the header files like this:
```
mesh_field.h: MeshField<FieldValue, MeshType> (abstract)
mesh_field_linear.h: MeshFieldLinear<FieldValue, MeshType> (concrete)
surface_mesh_field.h (new file)
    SurfaceMeshField = MeshField<_,SurfaceMesh<_>> (abstract)
    SurfaceMeshFieldLinear = MeshFieldLinear<_,SurfaceMesh<_>> (concrete)
volume_mesh_field.h
    VolumeMeshField = MeshField<_, VolumeMesh<_>> (abstract)
    VolumeMeshFieldLinear = MeshFieldLinear<_,VolumeMesh<_>> (concrete)
surface_mesh.h: SurfaceMesh<T>
volume_mesh.h: VolumeMesh<T>
```
And the drake_cc_library would look like this:
```
drake_cc_library("mesh_field")
    mesh_field.h
    mesh_field_linear.h
drake_cc_library("surface_mesh_field")
    surface_mesh.h
    surface_mesh_field.h
drake_cc_library("volume_mesh_field")
    volume_mesh.h
     volume_mesh_field.h
```
The current structure is slightly different.  Right now there is no `surface_mesh_field.h`, and the templates SurfaceMeshField and SurfaceMeshFieldLinear are in the files mesh_field.h, mesh_field_linear.h respectively.

This new packaging will change dependency in `contact_surface` in `geometry/query_results` from
```
drake_cc_library(name = "contact_surface", deps = "//geometry/proximity:mesh_field",)
```
to
```
drake_cc_library(name = "contact_surface", deps = "//geometry/proximity:surface_mesh_field",)
```

Other proposals are welcome too.

SeanCurtis-TRI, 2020-04-06:
An alternative resolution:

It isn't clear that the `MeshField` *family* is particularly providing value. 

1. We currently have no plan on actually using any implementation except the linear field. Even if we opt for a non-linear pressure field, it will still be represented with a piecewise linear field.
2. We've introduced methods to the linear implementation that doesn't exist for the base class (e.g., `CalcGradientField()`) because adding it to the base class was too painful. See [this discussion](https://reviewable.io/reviews/robotlocomotion/drake/12730#-M-z6Z_G7skCbxdAKCgb:-M-z6Z_G7skCbxdAKCgc:b92z066).

SeanCurtis-TRI, 2021-04-12:
I'm going to resolve this shortly by eliminating `MeshField` entirely. In my efforts to make hydroelastic contact differentiable, the class has proven to be an impediment. So, stay tuned for a PR to resolve this issue.

closed 2021-04-14 by SeanCurtis-TRI