RobotLocomotion/drake #11638: pybind11 build error under clang-6.0 --std=c++17

jwnimmer-tri, 2019-06-13:
From a `@drake-jenkins-bot linux-bionic-clang-bazel-experimental-cxx17-release please` CI build:
```
In file included from bazel-out/k8-opt/bin/bindings/pydrake/common/_virtual_includes/cpp_param_pybind/drake/bindings/pydrake/common/cpp_param_pybind.h:10:
external/pybind11/include/pybind11/pybind11.h:1005:9: error: no matching function for call to 'operator delete'
        ::operator delete(p, s, std::align_val_t(a));
        ^~~~~~~~~~~~~~~~~
```

Probably the same as pybind/pybind11#1604.

I wonder if we can fix it in our pybind11 fork, without having to add compiler flags?

jamiesnape, 2019-06-13:
My reading is that it is because we are using `clang-6` and `libstdc++` rather than `libc++`, `__STDCPP_DEFAULT_NEW_ALIGNMENT__` is defined by the compiler, without the necessary library support for `void operator delete(3)`. Checking `_LIBCPP_VERSION` and `__GLIBCXX__` in the fork may be a fix.

jamiesnape, 2019-06-14:
(e.g., `defined(_LIBCPP_VERSION) && _LIBCPP_VERSION >= 4000`)

jwnimmer-tri, 2019-06-14:
I have a fix ready.  It's actually `__cpp_sized_deallocation` that was the problem, not the `__cpp_aligned_new`.

jamiesnape, 2019-06-14:
Yes, I copied the wrong define. This commit: https://github.com/llvm/llvm-project/commit/0ae4f23fdcd8a278c58ea3ae2f062220763bcb66#diff-850e6a9fd4848551bd1419edb68bab55

PR #11650 (2019-06-14): pybind11: Update to latest master
Closes #11638.  Incorporates RobotLocomotion/pybind11#30.

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11650)
closed 2019-06-14 by jwnimmer-tri