RobotLocomotion/drake #11648: pydrake: Figure out why `py::bind_vector` does not like `AutoDiffXd`

EricCousineau-TRI, 2019-06-14:
See TODO here:
https://github.com/RobotLocomotion/drake/blob/f45733d57d491980375d6b359e14d4edd6e66ed1/bindings/pydrake/multibody/test/plant_test.py#L572-L576

With that code, the following happens:
```py
cls_f = VectorExternallyAppliedSpatialForced_[float]
cls_ad = VectorExternallyAppliedSpatialForced_[AutoDiffXd]
len(cls_f())  # Works
len(cls_ad())  # Fails with `ValueError: vector::reserve`
```

mntan3, 2020-05-12:
I believe this happens because we only have this call `PYBIND11_MAKE_OPAQUE(std::vector<drake::math::RigidTransform<double>>);` for the float type. I believe we can add a similar line for `AutoDiffXd` and it will work

nsinghal7, 2020-05-14:
I'm running into this issue with trying to do any access of a VectorExternallyAppliedSpatialForced_ of type AutoDiffXd or Expression. I believe it is this particular line that needs to be duplicated for the AutoDiffXd and Expression types: https://github.com/RobotLocomotion/drake/blob/28dff225df67a06f0e606d536183775453bad8de/bindings/pydrake/multibody/plant_py.cc#L25

mntan3, 2020-05-20:
Made a PR: https://github.com/RobotLocomotion/drake/pull/13359

closed 2020-05-20 by EricCousineau-TRI

EricCousineau-TRI, 2020-05-22:
FTR In convos with @jwnimmer-tri, and in writing a response in #13364, it would have been better to just bind it by value, rather than using an opaque class.

That being said, this current solution works, and thank y'all for identifying the root cause!