RobotLocomotion/drake #11649: Better place for python GUI example

guzhaoyuan, 2019-06-14:
[image]

https://github.com/RobotLocomotion/drake/blob/master/manipulation/util/geometry_inspector.py
is a GUI that we all love to use. But it is buried so deep that no one could find it. The usage example is in its comments (Difficult for user to use). 
I believe there should be an explicit example for this tool, maybe a README in the examples/ folder, or in the upcoming python tutorial.

RussTedrake, 2019-06-21:
i agree!  i was planning to include it (or a jupyter notebook equivalent) in my MBP tutorial that will land soonish on the drake main page.

RussTedrake, 2021-09-19:
Duplicate of #14980 .  (admittedly this one came first, but #14980 has the updated version of the resolution plan)

closed 2021-09-19 by RussTedrake