RobotLocomotion/drake #11651: InverseDynamicsController output confusing

guzhaoyuan, 2019-06-14:
I spent hours trying to use `InverseDynamicsController`. I finally figured out that the output of `InverseDynamicsController` is actually the generalized_force `Bu` and should be directed to `plant.get_applied_generalized_force_input_port()`. There is no warning or error when I connected to the `plant.get_actuation_input_port()` which only accept `u`, and no evidence in the documentation tells the user about the correct port.

There is no issue for fully actuated single branched robot, because `B` is identity matrix. So in the iiwa `InverseDynamicsController` example it is using `plant.get_actuation_input_port()` and totally fine.

The document for `InverseDynamicsController` output port
[image]

The document for `plant` input ports
[image]

We should detect the connection between wrong port and give error, add clear explanation about how to use `InverseDynamicsController` correctly, add correct examples.

SeanCurtis-TRI, 2019-06-19:
@edrumwri I'm assigning this to you -- feel free to pass it along if you feel you're the wrong assignee.

edrumwri, 2019-08-24:
The documentation for this class states that "this class assumes the robot is fully actuated..." and as you noted, "There is no issue for fully actuated ... robot"

closed 2019-08-24 by edrumwri