RobotLocomotion/drake #11655: vtk@8.1 image not found on mac

samlaf, 2019-06-16:
When I import pydrake I get

`Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/opt/drake/lib/python2.7/site-packages/pydrake/__init__.py", line 32, in <module>
    from . import common
  File "/usr/local/opt/drake/lib/python2.7/site-packages/pydrake/common/__init__.py", line 3, in <module>
    from ._module_py import *
ImportError: dlopen(/usr/local/opt/drake/lib/python2.7/site-packages/pydrake/common/_module_py.so, 2): Library not loaded: /usr/local/opt/vtk@8.1/lib/libvtkFiltersSources-8.1.1.dylib
  Referenced from: /usr/local/opt/drake/lib/python2.7/site-packages/pydrake/common/_module_py.so
  Reason: image not found`

vtk@8.1 is not installed, only vtk@8.2.
There is a line in drake's install_prereqs which I don't understand.
`# TODO(jamiesnape): Remove line uninstalling embree, ospray, and vtk@8.1
# formulae on or after 2019-07-01.
/usr/local/bin/brew uninstall --force embree ospray vtk@8.1`

This seems to try to uninstall vtk@8.1? But its not even installed, and actually it doesn't even seem available anymore on brew (only vtk@8.2 is found)

jamiesnape, 2019-06-17:
Did you rebuild drake and reinstall drake after running `install_prereqs.sh`?

samlaf, 2019-06-17:
Hmm possibly... I had installed it a long time ago, and might have reinstalled it lately without realizing it was already installed. :\
I'm guessing from your comment I shouldn't have, and that it might be causing my problem?
What's the best way to fix this? Uninstall everything and reinstall from scratch? Is there a clean way to do this (without me forgetting to uninstall some libraries/components/parts/etc)?

jamiesnape, 2019-06-17:
Easiest way is to remove the drake install and download https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-mac.tar.gz. You don't need to uninstall any dependencies.

samlaf, 2019-06-17:
Thank you! Everything's working now.
Turns out I actually had an old version of drake in /usr/local/opt which my pythonpath was linking to...

Even python3.7 version is working now. Perhaps you should update
https://drake.mit.edu/python_bindings.html#python-bindings-binary
which says:
[image]

and confused me (plus it's saying /usr/local).

jamiesnape, 2019-06-17:
Most of that box needs deleting, to be honest.

closed 2019-06-20 by jamiesnape