RobotLocomotion/drake #11657: ThreadSanitizer: lock-order-inversion (potential deadlock) when running snopt_solver_test with SNOPT/Fortran

jamiesnape, 2019-06-17:
For example:
```
 [ RUN      ] SnoptTest.TestPrintFile
 /media/ephemeral0/ubuntu/workspace/linux-bionic-clang-bazel-weekly-snopt-fortran-thread-sanitizer/_bazel_ubuntu/953efe98888d5cb3eea9c5e9e37e36f6/sandbox/linux-sandbox/3193/execroot/drake/_tmp/692dd5c2e8e365586fab68b6558c75db/snopt.out
 ==================
 WARNING: ThreadSanitizer: lock-order-inversion (potential deadlock) (pid=17)
   Cycle in lock order graph: M1135 (0x7fdd272000a0) => M2094 (0x7b58000006d8) => M1135
 
   Mutex M2094 acquired here while holding mutex M1135 in main thread:
     #0 pthread_mutex_lock ??:? (snopt_solver_test+0x456527)
     #1 _gfortran_transfer_complex128_write ??:? (libgfortran.so.4+0x196d18)
     #2 drake::solvers::(anonymous namespace)::SolveWithGivenOptions(drake::solvers::MathematicalProgram const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1> > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, int> > > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, double> > > const&, int*, double*, drake::EigenPtr<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, drake::solvers::SnoptSolverDetails*) /proc/self/cwd/solvers/snopt_solver.cc:782 (snopt_solver_test+0x89f0db)
     #3 drake::solvers::SnoptSolver::DoSolve(drake::solvers::MathematicalProgram const&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, drake::solvers::SolverOptions const&, drake::solvers::MathematicalProgramResult*) const /proc/self/cwd/solvers/snopt_solver.cc:1077 (snopt_solver_test+0x89ed8f)
     #4 drake::solvers::SolverBase::Solve(drake::solvers::MathematicalProgram const&, stx::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, stx::optional<drake::solvers::SolverOptions> const&, drake::solvers::MathematicalProgramResult*) const /proc/self/cwd/solvers/solver_base.cc:52 (snopt_solver_test+0xa06b8f)
     #5 drake::solvers::SolverBase::Solve(drake::solvers::MathematicalProgram const&, stx::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, stx::optional<drake::solvers::SolverOptions> const&) const /proc/self/cwd/solvers/solver_base.cc:28 (snopt_solver_test+0xa069aa)
     #6 drake::solvers::test::SnoptTest_TestPrintFile_Test::TestBody() /proc/self/cwd/solvers/test/snopt_solver_test.cc:143 (snopt_solver_test+0x4dcf8d)
     #7 void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /proc/self/cwd/external/gtest/googletest/src/gtest.cc:2443 (snopt_solver_test+0xc20547)
     #8 void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /proc/self/cwd/external/gtest/googletest/src/gtest.cc:2479 (snopt_solver_test+0xc0fe32)
     #9 testing::Test::Run() /proc/self/cwd/external/gtest/googletest/src/gtest.cc:2517 (snopt_solver_test+0xc00167)
     #10 testing::TestInfo::Run() /proc/self/cwd/external/gtest/googletest/src/gtest.cc:2693 (snopt_solver_test+0xc009e8)
     #11 testing::TestCase::Run() /proc/self/cwd/external/gtest/googletest/src/gtest.cc:2811 (snopt_solver_test+0xc00fc1)
     #12 testing::internal::UnitTestImpl::RunAllTests() /proc/self/cwd/external/gtest/googletest/src/gtest.cc:5177 (snopt_solver_test+0xc09e41)
     #13 bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /proc/self/cwd/external/gtest/googletest/src/gtest.cc:2443 (snopt_solver_test+0xc23837)
     #14 bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /proc/self/cwd/external/gtest/googletest/src/gtest.cc:2479 (snopt_solver_test+0xc11fc2)
     #15 testing::UnitTest::Run() /proc/self/cwd/external/gtest/googletest/src/gtest.cc:4786 (snopt_solver_test+0xc09b28)
     #16 RUN_ALL_TESTS() /proc/self/cwd/external/gtest/googletest/include/gtest/gtest.h:2341 (snopt_solver_test+0xbc883b)
     #17 main /proc/self/cwd/common/test_utilities/drake_cc_googletest_main.cc:32 (snopt_solver_test+0xbc8626)
 
   Mutex M1135 previously acquired by the same thread here:
     #0 pthread_mutex_lock ??:? (snopt_solver_test+0x456527)
     #1 _gfortran_transfer_complex128_write ??:? (libgfortran.so.4+0x196d9c)
     #2 drake::solvers::(anonymous namespace)::SolveWithGivenOptions(drake::solvers::MathematicalProgram const&, Eigen::Ref<Eigen::Matrix<double, -1, 1, 0, -1, 1> const, 0, Eigen::InnerStride<1> > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, int, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, int> > > const&, std::unordered_map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, double, std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::equal_to<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, double> > > const&, int*, double*, drake::EigenPtr<Eigen::Matrix<double, -1, 1, 0, -1, 1> >, drake::solvers::SnoptSolverDetails*) /proc/self/cwd/solvers/snopt_solver.cc:782 (snopt_solver_test+0x89f0db)
     #3 drake::solvers::SnoptSolver::DoSolve(drake::solvers::MathematicalProgram const&, Eigen::Matrix<double, -1, 1, 0, -1, 1> const&, drake::solvers::SolverOptions const&, drake::solvers::MathematicalProgramResult*) const /proc/self/cwd/solvers/snopt_solver.cc:1077 (snopt_solver_test+0x89ed8f)
     #4 drake::solvers::SolverBase::Solve(drake::solvers::MathematicalProgram const&, stx::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, stx::optional<drake::solvers::SolverOptions> const&, drake::solvers::MathematicalProgramResult*) const /proc/self/cwd/solvers/solver_base.cc:52 (snopt_solver_test+0xa06b8f)
     #5 drake::solvers::SolverBase::Solve(drake::solvers::MathematicalProgram const&, stx::optional<Eigen::Matrix<double, -1, 1, 0, -1, 1> > const&, stx::optional<drake::solvers::SolverOptions> const&) const /proc/self/cwd/solvers/solver_base.cc:28 (snopt_solver_test+0xa069aa)
[... cut ...] t_solver_test+0xc11fc2)
     #23 testing::UnitTest::Run() /proc/self/cwd/external/gtest/googletest/src/gtest.cc:4786 (snopt_solver_test+0xc09b28)
     #24 RUN_ALL_TESTS() /proc/self/cwd/external/gtest/googletest/include/gtest/gtest.h:2341 (snopt_solver_test+0xbc883b)
     #25 main /proc/self/cwd/common/test_utilities/drake_cc_googletest_main.cc:32 (snopt_solver_test+0xbc8626)
 
 SUMMARY: ThreadSanitizer: lock-order-inversion (potential deadlock) ??:? in pthread_mutex_lock
 ==================
```

jamiesnape, 2019-06-17:
I am probably going to just add `no_tsan` to the test unless someone has time to investigate. FYI @jwnimmer-tri.

jwnimmer-tri, 2019-06-17:
Last time I hunted down a potential race in `libgfortran.so` it was a false positive, and was even commented in the source code about why it was correct, but in a way that dynamic analysis could not infer.  I'm probably not going to look into this one anytime soon.

jwnimmer-tri, 2020-04-21:
Given my past experience noted above, I don't think this is worth chasing down.  If anyone experiences any deadlocks in production, we can dig in at that point.

closed 2020-04-21 by jwnimmer-tri