RobotLocomotion/drake #11668: pydrake: For core templated value types, ensure `cast[T]`, `ToType(T)`, or something else is exposed

EricCousineau-TRI, 2019-06-18:
Per @mpetersen94's Slack question:
https://drakedevelopers.slack.com/archives/C3YB3EV5W/p1560874172017900

> With `RigidTransform` now supporting Autodiff in python, what do I need to do to multiply a double-typed `RigidTransform` by a Autodiff-typed vector? Right now it seems to say that the `RigidTranform` needs to be the same type as the vector.

Current workaround would be to "explode" the type into its parts, and then do the conversion... but yuck. It should be simple enough to expose `cast[]` or what not.

mpetersen94, 2019-06-18:
Would binding [this method](https://drake.mit.edu/doxygen_cxx/classdrake_1_1math_1_1_rigid_transform.html#af22020b0567a392bfc275031a868657f) enable this?

EricCousineau-TRI, 2019-06-18:
Not exactly... All of those methods are templated on `T`, and thus constrains `other` to also be of type `T`.

Generally, in code when this needs to be done, `.cast<T>()` is used:
https://github.com/RobotLocomotion/drake/blob/a14a4cc82ae0572e82bf15033d0b23988ad8a414/geometry/proximity/test/distance_sphere_to_shape_test.cc#L146

mpetersen94, 2019-07-26:
Does this have an eta? Even a hacky version that isn't ready for drake master but can convert `RigidTranform` and `RotationMatrix` from `double` to `AutoDiffXd` would be helpful.

EricCousineau-TRI, 2019-07-26:
Took a whack at it in #11851. Lemme know how it looks.

PR #11851 (2019-07-29): py: Expose `.cast[T]()` for SE(3)-related values
Resolves #11668 (at least for immediately related types)

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11851)
closed 2019-07-29 by EricCousineau-TRI