RobotLocomotion/drake #11673: py multibody: Mach-O can't find `WeldJoint<Expression>::DoCloneToScalar(MultibodyTree<AutoDiffXd> const&) const`

EricCousineau-TRI, 2019-06-19:
Slack discussion:
https://drakedevelopers.slack.com/archives/C270MN28G/p1560950426013200

Relevant acute failure on CI:
https://drake-cdash.csail.mit.edu/testDetails.php?test=29601668&build=1184162

Demangling the symbol:
```
$ c++filt -_ -t __ZNK5drake9multibody9WeldJointINS_8symbolic10ExpressionEE15DoCloneToScalarERKNS0_8internal13MultibodyTreeIN5Eigen14AutoDiffScalarINS7_6MatrixIdLin1ELi1ELi0ELin1ELi1EEEEEEE
drake::multibody::WeldJoint<drake::symbolic::Expression>::DoCloneToScalar(drake::multibody::internal::MultibodyTree<Eigen::AutoDiffScalar<Eigen::Matrix<double, -1, 1, 0, -1, 1> > > const&) const
```

@jwnimmer-tri's suggestions:

> jwnimmer-tri  [1 hour ago]
> The first thing I'd do is try adding `~WeldJoint() final;` to the header and a definition _in the cc file_ and see if that helps the Mach-O linker find its pants.
> 
> jwnimmer-tri  [1 hour ago]
> Having `const std::string& type_name() const override` as the lexically first virtual method in the header file have _an inline definition_ is playing with fire.
> 
> jwnimmer-tri  [1 hour ago]
> Oh gosh, and it has a `never_destroyed` in it.  Double fire. (edited)
> 
> jwnimmer-tri  [1 hour ago]
> So then the second thing I'd try is to move the `type_name` definition to the cc file.

Relates revert, #11672

\cc @ggould-tri @SeanCurtis-TRI @m-chaturvedi

EricCousineau-TRI, 2019-06-19:
Er, it looks like the real culprit is that we just don't instantiate the weld joint for symbolic?
https://github.com/RobotLocomotion/drake/blob/8f87aabc46d8d750be4682457673dd8ffb99cf46/multibody/tree/weld_joint.cc#L62-L63

TBH, I have *no* idea why this was even passing on Linux???

EricCousineau-TRI, 2019-06-19:
My guess is that Linux didn't fail because it used a sort of lazy symbol loading, while Mac used something more immediate?

jwnimmer-tri, 2019-06-19:
The template could have been implicitly instantiated on Linux.  Just because the explicit instantiation was missing doesn't mean that the compiler is forbidden from implicitly instantiating it.  Or, the symbol was never used (inlined) on Linux but not macOS.  Or, ...

EricCousineau-TRI, 2019-06-19:
Well, the definition itself isn't available in the header file... And applying this diff on top of 50be4a3 doesn't even generate an error, so I'm completely stumped:
```patch
diff --git a/bindings/pydrake/multibody/test/plant_test.py b/bindings/pydrake/multibody/test/plant_test.py
index 0eadc9229..cb07c6734 100644
--- a/bindings/pydrake/multibody/test/plant_test.py
+++ b/bindings/pydrake/multibody/test/plant_test.py
@@ -360,7 +360,12 @@ class TestPlant(unittest.TestCase):
             A=plant_f.world_frame(), B=plant_f.GetFrameByName("base"))
         plant_f.Finalize()
         plant = to_type(plant_f, T)
+        print("---")
         print(plant)
+        if T != AutoDiffXd:
+            print(plant.ToAutoDiffXd())
+        if T != Expression:
+            print(plant.ToSymbolic())
         return
 
         context = plant.CreateDefaultContext()
```

However, I'm insufficiently interested to dig any further :P

jwnimmer-tri, 2019-06-19:
The implementation of `WeldJoint<double>::DoCloneToScalar(const internal::MultibodyTree<Expression>&) ` ends up doing `std::make_unique<WeldJoint<Expression>>` which implicitly instantiates `WeldJoint` on `Expression`.

EricCousineau-TRI, 2019-06-19:
Ohhhhh, that makes sense. Thanks!

PR #11674 (2019-06-19): multibody: Ensure WeldJoint is instantiated for Expression
Resolves #11673

Supersedes #11672

\cc @ggould-tri @jwnimmer-tri

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11674)
closed 2019-06-19 by jwnimmer-tri