RobotLocomotion/drake #11692: ci: Timeline to drop Mac High Sierra support

EricCousineau-TRI, 2019-06-20:
To reduce number of builds, we may want to turn down on Mac by turning of High Sierra.

Any hard votes against doing this by September, 2019? (in preparation for Mac Catalina)

Relates #11610 

\cc @jwnimmer-tri @RussTedrake @soonho-tri

jwnimmer-tri, 2019-06-20:
In my opinion we should turn off High Sierra CI configurations exactly when we begin adding Catalina CI configurations, per our de-facto {N, N-1} policy for macOS.

soonho-tri, 2019-06-20:
- Apple says that "macOS Catalina (10.15) will be released in the fall of 2019" without further details.
 - I guess it will take some time for our CI infrastructures to be ready to run 10.15. (as it happened last year). So I expect that we will have 2 - 3 months that we will only run 10.14 in CI if we drop 10.13.

So, if we want to stick to the N/N-1 policy for macOS, we need to wait until 10.15 is available in CI. I'm fine if we want to change the policy temporarily just to supporting 10.14.

jwnimmer-tri, 2019-06-20:
Let me clarify my position in light of that: we should turn off High Sierra CI configurations exactly when Catalina is released, or we begin adding Catalina beta CI configurations, whichever comes first.

jamiesnape, 2019-06-20:
My guess would be that Catalina builds will be up fairly quickly (within a month of release). Mojave was unusual since the required filesystem type changed.

RussTedrake, 2019-06-21:
fwiw -- i have no attachment to high sierra.  all the mac use cases I'm aware of are on Mojave.

EricCousineau-TRI, 2019-09-09:
Relates #12003; from that discussion, and the Slack poll, seems like no one really needs Python 2 on Mac, and Homebrew's dropping of Python 2 support means that we may want to drop this as well.

Updated the checkbox in #10606 to xref this issue.

jamiesnape, 2019-10-11:
Note that we are waiting on `clp`, `diffstat`, `doxygen`, and `llvm@6` dependencies to be built for Catalina for unprovisioned builds to be switched on.

EricCousineau-TRI, 2019-10-31:
Marking this as resolved for now; will create separate issue about Catalina.

closed 2019-10-31 by EricCousineau-TRI