RobotLocomotion/drake #11695: //examples/schunk_wsg:schunk_wsg_lift_test: Invalid read of size 8

jamiesnape, 2019-06-20:
```
FAIL: //examples/schunk_wsg:schunk_wsg_lift_test
INFO: From Testing //examples/schunk_wsg:schunk_wsg_lift_test:
==================== Test output for //examples/schunk_wsg:schunk_wsg_lift_test:
==17== Memcheck, a memory error detector
==17== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==17== Using Valgrind-3.13.0 and LibVEX; rerun with -h for copyright info
==17== Command: examples/schunk_wsg/schunk_wsg_lift_test
==17== 
Using drake_cc_googletest_main.cc

[==========] Running 2 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 2 tests from CompliantAndTimeSteppingTest/SchunkWsgLiftTest
[ RUN      ] CompliantAndTimeSteppingTest/SchunkWsgLiftTest.BoxLiftTest/0
[2019-06-20 14:04:55.444] [console] [info] Adding joint lift_joint to the plant.
[2019-06-20 14:04:55.487] [console] [info] Adding joint x_translate_joint to the plant.
[2019-06-20 14:04:55.489] [console] [info] Adding joint y_translate_joint to the plant.
[2019-06-20 14:04:55.563] [console] [info] Adding joint left_finger_sliding_joint to the plant.
[2019-06-20 14:04:55.564] [console] [info] Adding joint right_finger_sliding_joint to the plant.
==17== Invalid read of size 8
==17==    at 0x1AD20A: coeff (CoreEvaluators.h:840)
==17==    by 0x1AD20A: coeff (DenseCoeffsBase.h:144)
==17==    by 0x1AD20A: operator[] (DenseCoeffsBase.h:163)
==17==    by 0x1AD20A: drake::examples::schunk_wsg::(anonymous namespace)::SchunkWsgLiftTest_BoxLiftTest_Test::TestBody() (schunk_wsg_lift_test.cc:455)
==17==    by 0x5C616C: HandleSehExceptionsInMethodIfSupported<testing::Test, void> (gtest.cc:2443)
==17==    by 0x5C616C: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2479)
==17==    by 0x5C63A1: testing::Test::Run() (gtest.cc:2517)
==17==    by 0x5C66A7: testing::TestInfo::Run() (gtest.cc:2693)
==17==    by 0x5C6936: testing::TestCase::Run() (gtest.cc:2811)
==17==    by 0x5C6DCF: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:5177)
==17==    by 0x5C6F1C: HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (gtest.cc:2443)
==17==    by 0x5C6F1C: bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) (gtest.cc:2479)
==17==    by 0x5C7125: testing::UnitTest::Run() (gtest.cc:4786)
==17==    by 0x18CA64: RUN_ALL_TESTS (gtest.h:2341)
==17==    by 0x18CA64: main (drake_cc_googletest_main.cc:32)
==17==  Address 0xb3ec4d8 is 0 bytes after a block of size 184 alloc'd
==17==    at 0x4C2FB0F: malloc (vg_replace_malloc.c:299)
==17==    by 0x585628: aligned_malloc (Memory.h:159)
==17==    by 0x585628: conditional_aligned_malloc<true> (Memory.h:214)
==17==    by 0x585628: conditional_aligned_new_auto<double, true> (Memory.h:374)
==17==    by 0x585628: resize (DenseStorage.h:557)
==17==    by 0x585628: resize (PlainObjectBase.h:293)
==17==    by 0x585628: resize_if_allowed<Eigen::Matrix<double, -1, 1>, Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1> >, double, double> (AssignEvaluator.h:720)
==17==    by 0x585628: call_dense_assignment_loop<Eigen::Matrix<double, -1, 1>, Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1> >, Eigen::internal::assign_op<double, double> > (AssignEvaluator.h:734)
==17==    by 0x585628: run (AssignEvaluator.h:879)
==17==    by 0x585628: call_assignment_no_alias<Eigen::Matrix<double, -1, 1>, Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1> >, Eigen::internal::assign_op<double, double> > (AssignEvaluator.h:836)
==17==    by 0x585628: _set_noalias<Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1> > > (PlainObjectBase.h:728)
==17==    by 0x585628: _init1<Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1> >, Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1> > > (PlainObjectBase.h:812)
==17==    by 0x585628: Matrix<Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1> > > (Matrix.h:296)
==17==    by 0x585628: BasicVector (basic_vector.h:40)
==17==    by 0x585628: DoClone (basic_vector.h:142)
==17==    by 0x585628: drake::systems::BasicVector<double>::Clone() const (basic_vector.h:119)
==17==    by 0x193772: to_storage (value.h:664)
==17==    by 0x193772: drake::Value<drake::systems::BasicVector<double> >::Value(drake::systems::BasicVector<double> const&) (value.h:773)
==17==    by 0x1937CB: make_unique<drake::Value<drake::systems::BasicVector<double> >, const drake::systems::BasicVector<double>&> (unique_ptr.h:825)
==17==    by 0x1937CB: drake::Value<drake::systems::BasicVector<double> >::Clone() const (value.h:839)
==17==    by 0x18D627: operator() (leaf_system.h:2744)
==17==    by 0x18D627: std::_Function_handler<std::unique_ptr<drake::AbstractValue, std::default_delete<drake::AbstractValue> > (), std::function<std::unique_ptr<drake::AbstractValue, std::default_delete<drake::AbstractValue> > ()> drake::systems::LeafSystem<double>::MakeAllocCallback<drake::systems::BasicVector<double> >(drake::systems::BasicVector<double> const&)::{lambda()#1}>::_M_invoke(std::_Any_data const&) (std_function.h:302)
==17==    by 0x570013: operator() (std_function.h:706)
==17==    by 0x570013: drake::systems::CacheEntry::Allocate() const (cache_entry.cc:38)
==17==    by 0x554664: drake::systems::LeafOutputPort<double>::DoAllocate() const (leaf_output_port.h:85)
==17==    by 0x56CCCB: drake::systems::OutputPort<double>::Allocate() const (output_port.h:148)
==17==    by 0x520154: drake::systems::DiagramOutputPort<double>::DoAllocate() const (diagram_output_port.h:98)
==17==    by 0x5658F3: Allocate (output_port.h:148)
==17==    by 0x5658F3: drake::systems::System<double>::AllocateOutput() const (system.h:131)
==17==    by 0x1AD014: drake::examples::schunk_wsg::(anonymous namespace)::SchunkWsgLiftTest_BoxLiftTest_Test::TestBody() (schunk_wsg_lift_test.cc:429)
==17==    by 0x5C616C: HandleSehExceptionsInMethodIfSupported<testing::Test, void> (gtest.cc:2443)
==17==    by 0x5C616C: void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) (gtest.cc:2479)
==17==    by 0x5C63A1: testing::Test::Run() (gtest.cc:2517)
==17==    by 0x5C66A7: testing::TestInfo::Run() (gtest.cc:2693)
==17==    by 0x5C6936: testing::TestCase::Run() (gtest.cc:2811)
==17==    by 0x5C6DCF: testing::internal::UnitTestImpl::RunAllTests() (gtest.cc:5177)
==17== 
[       OK ] CompliantAndTimeSteppingTest/SchunkWsgLiftTest.BoxLiftTest/0 (1392440 ms)
[ RUN      ] CompliantAndTimeSteppingTest/SchunkWsgLiftTest.BoxLiftTest/1
[2019-06-20 14:28:07.620] [console] [info] Adding joint lift_joint to the plant.
[2019-06-20 14:28:07.622] [console] [info] Adding joint x_translate_joint to the plant.
[2019-06-20 14:28:07.624] [console] [info] Adding joint y_translate_joint to the plant.
[2019-06-20 14:28:07.646] [console] [info] Adding joint left_finger_sliding_joint to the plant.
[2019-06-20 14:28:07.647] [console] [info] Adding joint right_finger_sliding_joint to the plant.
[       OK ] CompliantAndTimeSteppingTest/SchunkWsgLiftTest.BoxLiftTest/1 (2087155 ms)
[----------] 2 tests from CompliantAndTimeSteppingTest/SchunkWsgLiftTest (3479607 ms total)

[----------] Global test environment tear-down
[==========] 2 tests from 1 test case ran. (3479639 ms total)
[  PASSED  ] 2 tests.
==17== 
==17== HEAP SUMMARY:
==17==     in use at exit: 59,414 bytes in 173 blocks
==17==   total heap usage: 88,097,618 allocs, 88,097,445 frees, 57,848,313,528 bytes allocated
==17== 
==17== LEAK SUMMARY:
==17==    definitely lost: 0 bytes in 0 blocks
==17==    indirectly lost: 0 bytes in 0 blocks
==17==      possibly lost: 0 bytes in 0 blocks
==17==    still reachable: 59,414 bytes in 173 blocks
==17==         suppressed: 0 bytes in 0 blocks
==17== Reachable blocks (those to which a pointer was found) are not shown.
==17== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==17== 
==17== For counts of detected and suppressed errors, rerun with: -v
==17== ERROR SUMMARY: 2 errors from 1 contexts (suppressed: 0 from 0)
================================================================================
Target //examples/schunk_wsg:schunk_wsg_lift_test up-to-date:
  bazel-bin/examples/schunk_wsg/schunk_wsg_lift_test
INFO: Elapsed time: 3601.753s, Critical Path: 3584.56s
INFO: 1 process: 1 linux-sandbox.
INFO: Build completed, 1 test FAILED, 3 total actions
//examples/schunk_wsg:schunk_wsg_lift_test                               FAILED in 3580.7s
```

jamiesnape, 2019-06-20:
FYI @sammy-tri (from `git blame`)

sammy-tri, 2019-06-20:
That debug print is obviously wrong (the plant has a quaternion joint and so there will be more positions than velocities).  I'd just delete the print statements and the loop entirely (or go ahead and reassign to me).

commit 28b67447 (2019-06-21): Remove debug print which was reading off the end of the vector.

Fixes #11695
closed 2019-06-21 by sherm1