RobotLocomotion/drake #11699: py mp: `result.GetSolution(var)` segfaults if `not result.is_sucess()`

EricCousineau-TRI, 2019-06-20:
From @sadraddini's comment here:
https://github.com/RobotLocomotion/drake/issues/9633#issuecomment-497437906

I can't see this C++ (at least for a given solver), but I can see it in Python (at least for this given formulation).

TODO: Add repro code

EricCousineau-TRI, 2019-06-20:
@hongkai-dai Seems like this is a C++ issue in general (see edit above)

EricCousineau-TRI, 2019-06-20:
As Hongkai mentioned in upstream thread, it seems like it *might* be dependent on what solver it is.

EricCousineau-TRI, 2019-06-20:
Checking in debug mode, failure is caught using `DRAKE_ASSERT`:
```
abort: Failure at solvers/mathematical_program_result.cc:52 in GetVariableValue(): condition 'variable_values.rows() == static_cast<int>(variable_index->size())' failed.
```
Should we use `DRAKE_DEMAND` here (not to resolve this, but to make error more clear)?

That being said, my initial guess is that OSQP's code paths do not call `MathematicalProgramResult::set_x_val`, which seems to be confirmed here:
https://github.com/RobotLocomotion/drake/blob/3ee9090108b50e75ce88f259255ca9a8a08d11a1/solvers/osqp_solver.cc#L289-L319

Question is:
Should `x_val_` always be populated correctly, and if so, what should it be? (NaN?)
Or should `GetSolution(...)` fail-fast if it was unsuccessful? (thus avoiding the need for this policy?)

\cc @jwnimmer-tri

hongkai-dai, 2019-06-20:
I think when I call `MPResult::GetSolution()`, it should check if `x_val_` is empty or not. If `x_val_` is empty, then `MPResult::GetSolution()` should just return some NaNs, to indicate that the problem is not solved successfully.

I will send a quick PR.

EricCousineau-TRI, 2019-06-20:
Sounds good! Will reassign for now, thanks!

PR #11707 (2019-06-24): Set the x_val and optimal cost to nan by default in MathematicalProgramResult
Fixes #11699

---
This change is [[image]](https://reviewable.io/reviews/robotlocomotion/drake/11707)
closed 2019-06-24 by hongkai-dai