RobotLocomotion/drake #11700: `RollPitchYaw` performs `if` logic on symbolic

ggould-tri, 2019-06-20:
In `roll_pitch_yaw.cc`:185 there is an `if` on comparisons with T type variables (which can be symbolic).  This triggers an Evaluate without a meaningful environment and so blows up in my use case (random distributions over angles).

Those comparisons should use `if_then_else` instead, and this case should be tested.

See https://drakedevelopers.slack.com/conversation/C2WBPQDB7/p1561056705064800 for original conversation.

Thanks!

sherm1, 2019-06-20:
Looks like a job for you, @mitiguy !

sherm1, 2019-06-20:
@ggould-tri what's an appropriate priority for this?

ggould-tri, 2019-06-20:
I was able to work around it in my code by reordering my operations, so it's not a "things are broken" priority.  I think the question is how it relates to dynamics work.

The fix should be trivial; the test coverage I don't know.

jwnimmer-tri, 2020-12-09:
I believe that #10495 already resolved this.

closed 2020-12-09 by jwnimmer-tri