RobotLocomotion/drake #11710: //examples/kuka_iiwa_arm:dual_kuka_simulation_test: Invalid reads of size 8

jamiesnape, 2019-06-21:
```
==================== Test output for //examples/kuka_iiwa_arm:dual_kuka_simulation_test:

Invalid read of size 8
   at 0x18F5D8: coeff (CoreEvaluators.h:1206)
   by 0x18F5D8: assignCoeff (AssignEvaluator.h:631)
   by 0x18F5D8: assignCoeffByOuterInner (AssignEvaluator.h:645)
   by 0x18F5D8: run (AssignEvaluator.h:326)
   by 0x18F5D8: void Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<double, -1, 1, 0, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, -1>, Eigen::internal::assign_op<double, double> >(Eigen::Matrix<double, -1, 1, 0, -1, 1>&, Eigen::Replicate<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, -1> const&, Eigen::internal::assign_op<double, double> const&) (AssignEvaluator.h:741)
   by 0x1A072F: run (AssignEvaluator.h:879)
   by 0x1A072F: call_assignment_no_alias<Eigen::Matrix<double, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1>, Eigen::internal::assign_op<double, double> > (AssignEvaluator.h:836)
   by 0x1A072F: call_assignment<Eigen::Matrix<double, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1>, Eigen::internal::assign_op<double, double> > (AssignEvaluator.h:804)
   by 0x1A072F: call_assignment<Eigen::Matrix<double, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1> > (AssignEvaluator.h:782)
   by 0x1A072F: _set<Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1> > (PlainObjectBase.h:710)
   by 0x1A072F: operator=<Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1> > (Matrix.h:225)
   by 0x1A072F: drake::examples::kuka_iiwa_arm::(anonymous namespace)::DoMain() (kuka_simulation.cc:106)
   by 0x1848BD: main (kuka_simulation.cc:231)
 Address 0xc107e50 is 0 bytes inside a block of size 56 free'd
   at 0x4C30D3B: free (vg_replace_malloc.c:530)
   by 0x18AE36: aligned_free (Memory.h:177)
   by 0x18AE36: conditional_aligned_free<true> (Memory.h:230)
   by 0x18AE36: conditional_aligned_delete_auto<double, true> (Memory.h:416)
   by 0x18AE36: resize (DenseStorage.h:555)
   by 0x18AE36: Eigen::PlainObjectBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> >::resize(long, long) (PlainObjectBase.h:293)
   by 0x18F5BF: resize_if_allowed<Eigen::Matrix<double, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1>, double, double> (AssignEvaluator.h:720)
   by 0x18F5BF: void Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<double, -1, 1, 0, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, -1>, Eigen::internal::assign_op<double, double> >(Eigen::Matrix<double, -1, 1, 0, -1, 1>&, Eigen::Replicate<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, -1> const&, Eigen::internal::assign_op<double, double> const&) (AssignEvaluator.h:734)
   by 0x1A072F: run (AssignEvaluator.h:879)
   by 0x1A072F: call_assignment_no_alias<Eigen::Matrix<double, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1>, Eigen::internal::assign_op<double, double> > (AssignEvaluator.h:836)
   by 0x1A072F: call_assignment<Eigen::Matrix<double, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1>, Eigen::internal::assign_op<double, double> > (AssignEvaluator.h:804)
   by 0x1A072F: call_assignment<Eigen::Matrix<double, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1> > (AssignEvaluator.h:782)
   by 0x1A072F: _set<Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1> > (PlainObjectBase.h:710)
   by 0x1A072F: operator=<Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1> > (Matrix.h:225)
   by 0x1A072F: drake::examples::kuka_iiwa_arm::(anonymous namespace)::DoMain() (kuka_simulation.cc:106)
   by 0x1848BD: main (kuka_simulation.cc:231)
 Block was alloc'd at
   at 0x4C2FB0F: malloc (vg_replace_malloc.c:299)
   by 0x1A5A05: aligned_malloc (Memory.h:159)
   by 0x1A5A05: conditional_aligned_malloc<true> (Memory.h:214)
   by 0x1A5A05: conditional_aligned_new_auto<double, true> (Memory.h:374)
   by 0x1A5A05: resize (DenseStorage.h:557)
   by 0x1A5A05: resize (PlainObjectBase.h:319)
   by 0x1A5A05: drake::examples::kuka_iiwa_arm::SetPositionControlledIiwaGains(Eigen::Matrix<double, -1, 1, 0, -1, 1>*, Eigen::Matrix<double, -1, 1, 0, -1, 1>*, Eigen::Matrix<double, -1, 1, 0, -1, 1>*) (iiwa_common.cc:111)
   by 0x1A06FD: drake::examples::kuka_iiwa_arm::(anonymous namespace)::DoMain() (kuka_simulation.cc:105)
   by 0x1848BD: main (kuka_simulation.cc:231)

Invalid read of size 8
   at 0x18F5D8: coeff (CoreEvaluators.h:1206)
   by 0x18F5D8: assignCoeff (AssignEvaluator.h:631)
   by 0x18F5D8: assignCoeffByOuterInner (AssignEvaluator.h:645)
   by 0x18F5D8: run (AssignEvaluator.h:326)
   by 0x18F5D8: void Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<double, -1, 1, 0, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, -1>, Eigen::internal::assign_op<double, double> >(Eigen::Matrix<double, -1, 1, 0, -1, 1>&, Eigen::Replicate<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, -1> const&, Eigen::internal::assign_op<double, double> const&) (AssignEvaluator.h:741)
   by 0x1A076B: run (AssignEvaluator.h:879)
   by 0x1A076B: call_assignment_no_alias<Eigen::Matrix<double, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1>, Eigen::internal::assign_op<double, double> > (AssignEvaluator.h:836)
   by 0x1A076B: call_assignment<Eigen::Matrix<double, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1>, Eigen::internal::assign_op<double, double> > (AssignEvaluator.h:804)
   by 0x1A076B: call_assignment<Eigen::Matrix<double, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1> > (AssignEvaluator.h:782)
   by 0x1A076B: _set<Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1> > (PlainObjectBase.h:710)
   by 0x1A076B: operator=<Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1> > (Matrix.h:225)
   by 0x1A076B: drake::examples::kuka_iiwa_arm::(anonymous namespace)::DoMain() (kuka_simulation.cc:107)
   by 0x1848BD: main (kuka_simulation.cc:231)
 Address 0xc107ed0 is 0 bytes inside a block of size 56 free'd
   at 0x4C30D3B: free (vg_replace_malloc.c:530)
   by 0x18AE36: aligned_free (Memory.h:177)
   by 0x18AE36: conditional_aligned_free<true> (Memory.h:230)
   by 0x18AE36: conditional_aligned_delete_auto<double, true> (Memory.h:416)
   by 0x18AE36: resize (DenseStorage.h:555)
   by 0x18AE36: Eigen::PlainObjectBase<Eigen::Matrix<double, -1, 1, 0, -1, 1> >::resize(long, long) (PlainObjectBase.h:293)
   by 0x18F5BF: resize_if_allowed<Eigen::Matrix<double, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1>, double, double> (AssignEvaluator.h:720)
   by 0x18F5BF: void Eigen::internal::call_dense_assignment_loop<Eigen::Matrix<double, -1, 1, 0, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, -1>, Eigen::internal::assign_op<double, double> >(Eigen::Matrix<double, -1, 1, 0, -1, 1>&, Eigen::Replicate<Eigen::Matrix<double, -1, 1, 0, -1, 1>, -1, -1> const&, Eigen::internal::assign_op<double, double> const&) (AssignEvaluator.h:734)
   by 0x1A076B: run (AssignEvaluator.h:879)
   by 0x1A076B: call_assignment_no_alias<Eigen::Matrix<double, -1, 1>, Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1>, Eigen::internal::assign_op<double, double> > (AssignEvaluator.h:836)
[... 44 lines cut ...]
   by 0x1A079D: _set<Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1> > (PlainObjectBase.h:710)
   by 0x1A079D: operator=<Eigen::Replicate<Eigen::Matrix<double, -1, 1>, -1, -1> > (Matrix.h:225)
   by 0x1A079D: drake::examples::kuka_iiwa_arm::(anonymous namespace)::DoMain() (kuka_simulation.cc:108)
   by 0x1848BD: main (kuka_simulation.cc:231)
 Block was alloc'd at
   at 0x4C2FB0F: malloc (vg_replace_malloc.c:299)
   by 0x1A5ADA: aligned_malloc (Memory.h:159)
   by 0x1A5ADA: conditional_aligned_malloc<true> (Memory.h:214)
   by 0x1A5ADA: conditional_aligned_new_auto<double, true> (Memory.h:374)
   by 0x1A5ADA: resize (DenseStorage.h:557)
   by 0x1A5ADA: resize (PlainObjectBase.h:293)
   by 0x1A5ADA: resize_if_allowed<Eigen::Matrix<double, -1, 1>, Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1> >, double, double> (AssignEvaluator.h:720)
   by 0x1A5ADA: call_dense_assignment_loop<Eigen::Matrix<double, -1, 1>, Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1> >, Eigen::internal::assign_op<double, double> > (AssignEvaluator.h:734)
   by 0x1A5ADA: run (AssignEvaluator.h:879)
[... cut ...]
   by 0x1A5ADA: operator=<Eigen::CwiseNullaryOp<Eigen::internal::scalar_constant_op<double>, Eigen::Matrix<double, -1, 1> > > (Matrix.h:225)
   by 0x1A5ADA: drake::examples::kuka_iiwa_arm::SetPositionControlledIiwaGains(Eigen::Matrix<double, -1, 1, 0, -1, 1>*, Eigen::Matrix<double, -1, 1, 0, -1, 1>*, Eigen::Matrix<double, -1, 1, 0, -1, 1>*) (iiwa_common.cc:118)
   by 0x1A06FD: drake::examples::kuka_iiwa_arm::(anonymous namespace)::DoMain() (kuka_simulation.cc:105)
   by 0x1848BD: main (kuka_simulation.cc:231)
```

`git blame` appears to suggest @pvarin and @sammy-tri.

sammy-tri, 2019-06-21:
Is there a bazel command line to reproduce this locally?

jamiesnape, 2019-06-21:
`bazel test --config=memcheck --test_lang_filters= //examples/kuka_iiwa_arm:dual_kuka_simulation_test`

sammy-tri, 2019-06-21:
Thanks!  Unless you're already working on this, go ahead and assign it to me and I'll batch it with the fix to #11695

commit 29f97073 (2019-06-21): Call eval() on the output of replicate before assigning to the thing which was replicated.

Fixes #11710
closed 2019-06-21 by sherm1