RobotLocomotion/drake #11721: Geometry names aren't displayed in drake_visualizer

SeanCurtis-TRI, 2019-06-25:
# Problem statement

When a geometry is registered with a specific name, one would expect it to appear applied to the geometry in `drake_visualizer`. That doesn't appear to be happening. Primitives are generally given a name based on an enumeration of the frame the geometry is fixed to. Some meshes are picking up names based on the obj.

SeanCurtis-TRI, 2019-06-25:
cc @hongkai-dai

hongkai-dai, 2019-06-25:
In MultibodyPlant, I registered many visual geometries (both box primitives, and mesh with obj files), each with a distinct name
```C++
 plant->RegisterVisualGeometry(
      plant->world_body(), X_WB, box, name,
      drake::geometry::IllustrationProperties(),
      scene_graph_);
```
I then publish the visual to drake_visualizer through
```C++
  drake::geometry::ConnectDrakeVisualizer(&builder_, *scene_graph_);

  diagram_ = builder_.Build();
  drake::systems::Simulator<double>(*diagram_).Initialize();

  // Create and initialize the diagram context.

  diagram_.Publish(diagram_context_);
```
But in Drake visualizer, none of these geometries have the name I registered, as shown in the screenshot
[image]

In drake-visualizer, all the box primitives (which are added to the world frame) are named "world geometry data", and the mesh obj (in the iiwa link) are named "Source_2::iiwa::iiwa_link_X geometry data"

SeanCurtis-TRI, 2019-07-31:
After some slight investigation the current state of affairs is as follows:

  1. There is no way to communicate geometry-specific names via the LCM interface.
  2. The current behavior names every geometry based on the simple rule: `[frame] geometry data` such that in the example above, the geometries shown are anchored geometries attached to the world.

The solution to this problem will require three coordinated efforts:

1. lcmt_viewer_geometry_data.lcm will need to include a "name" field (that it currently doesn't have).
2. `drake_visualizer` will need to consume that field.
3. `geometry_visualization.cc` will need to *set* the name into the field. 

The last point is easy and cheap. The first two items represent the majority of the work.

hongkai-dai, 2019-07-31:
We will reopen this issue if it becomes important.

closed 2019-07-31 by hongkai-dai