RobotLocomotion/drake #11725: Need a function to compute the translational velocity of a point attached to a body

hongkai-dai, 2019-06-25:
Currently we have a function `Frame::CalcSpatialVelocity` to compute the spatial velocity of a frame, but we don't have the function to compute the translational velocity of a point attached to a frame (I can compute it as v + omega.cross(p), but an API would be nicer).

amcastro-tri, 2019-07-01:
@hongkai-dai, you can use `SpatialVelocity::Shift()` which accomplishes exactly what you want. Please take a look at its docs and let me know if it solves your problem

hongkai-dai, 2019-07-01:
@amcastro-tri, that's what I did in my dev branch. But I would say it is not very convenient, especially if the  expressed frame of the shift vector `p_BpBq_E` is different from `B`. When that happens, I need to first compute the pose of `B` relative to `E`, then transform `p_BpBq_B` to `p_BpBq_E`, and then do the shift. It would be more convenient for the user, if there is a function `CalcPointsTranslationalVelocities()`, as a counter part of the existing function `CalcPointsPositions()`.

Anyway, this is not urgent. I called Shift function, just as you suggested. But it would be nicer to provide the API.

jwnimmer-tri, 2024-03-20:
All valid points, but nobody is planning to work on this, so "not planned".

closed 2024-03-20 by jwnimmer-tri