RobotLocomotion/drake #11742: Issues with installing DRAKE with matlab bindings

VishnuDevA7, 2019-06-28:
I am trying to install DRAKE toolbox with the "last_sha_with_original_matlab" tag (substantial MATLAB code) as per the instructions at https://drake.mit.edu/matlab_bindings.html for installing the MATLAB bindings. I run into errors such as:
"In file included from /home/dev/drake/drake/common/call_matlab.cc:1:0:
/home/dev/drake/build/../drake/common/call_matlab.h:7:40: fatal error: drake/common/matlab_rpc.pb.h: No such file or directory
compilation terminated".

I did find the file in /drake/build/common/ while it was searched for in /drake/drake/common. So I fixed this problem by making a copy of the same file in /drake/build/common. Then I ran into a lot of similar errors, first among them is :

"In file included from /home/dev/drake/build/../drake/common/monomial.h:16:0,
                 from /home/dev/drake/drake/common/monomial.cc:1:
/home/dev/drake/build/../drake/common/symbolic_expression.h:768:8: error: ‘ScalarBinaryOpTraits’ is not a class template
 struct ScalarBinaryOpTraits<drake::symbolic::Variable," 

I am building the bindings in the right way ? IF yes, how to fix these issues ?

RussTedrake, 2019-06-29:
I'm afraid that although we continue to maintain that tag, we do not officially support it.  (I actually open that sha to read some old code even still, but it's been a long time since i've tried to compile/run, and I'm not surprised that you might have trouble).  

I highly recommend using the latest versions, which will be up to date with all of your system libraries, etc.  If you want to use it from matlab, i recommend using pydrake with matlab's support for python: https://www.mathworks.com/help/matlab/call-python-libraries.html

I think it's time to clean up our sphinx documentation in this regard.  I will open a new issue for it.

closed 2019-06-29 by RussTedrake

VishnuDevA7, 2019-06-30:
Is it possible to let me know which version of Eigen3 library was used at the time of the corresponding tag?

jamiesnape, 2019-07-01:
Eigen: https://github.com/RobotLocomotion/drake/blob/d7f3c011d37d471d7b9293ecf2066c98d88b2a05/CMakeLists.txt#L35

Protobuf: https://github.com/protocolbuffers/protobuf/tree/efec7571047c680e11354bd30bf8bd3dfda15ff0 via https://github.com/RobotLocomotion/drake/tree/last_sha_with_original_matlab/externals