RobotLocomotion/drake #11745: [render] Assess viability of filament as PBR real-time RenderEngine implementation

SeanCurtis-TRI, 2019-06-28:
[Google filament](https://github.com/google/filament) is an open-source, cross-platform physically-based render engine (PBR). Images such as [this](https://google.github.io/filament/webgl/parquet.html) are highly suggestive that the render *quality* is what we're looking for. However, there are a number of issues that need to be considered:

1. Integration
    - Are there issues with exercising it Drake?
    - Can it support the `RenderEngine` API? E.g., can we get label, depth, and color images from it?
2. Mac-compatible
    - Although billed a cross-platform compatible, this needs to be confirmed for Drake.
3. Authoring 
    - What will it take to author content? 
    - What will it take to load content?
 
(Edit this as more issues become interesting.)

## Deliverable

Ultimately, this won't necessarily produce a finished implementation. Some test code and a document detailing the pros/cons of filament is the concrete deliverable.

SeanCurtis-TRI, 2019-06-28:
See @gizatt's initial analysis in the slack channel: https://drakedevelopers.slack.com/archives/C3GG2R4JY/p1543862958012700?thread_ts=1543862958.012700

soonho-tri, 2019-06-28:
> Mac-compatible

This might be useful: https://github.com/dartsim/homebrew-dart/blob/master/Formula/filament.rb

SeanCurtis-TRI, 2019-06-28:
Heh...looks like Amazon must be interested in it as well. (Ironic since they have their own game engine.)

SeanCurtis-TRI, 2020-05-13:
Closing this for now; we can revisit it if filament evolves.

closed 2020-05-13 by SeanCurtis-TRI