RobotLocomotion/drake #11746: [render] Extend parsers to include data for the perception role.

SeanCurtis-TRI, 2019-06-28:
## Problem statement

In `RenderEngineVtk`, materials for objects are generated in a really hackneyed way:

  - Only diffuse color can be specified (all other material parameters are defaulted).
  - If an OBJ named `path/to/bob.obj` is referenced *and* the file `path/to/bob.png` exists, that *mesh* will receive the image as a texture.

Furthermore, future renderers (e.g., OSPRay -- see #11744) have a completely different set of render parameters.

Ideally, we should extend the sdf/urdf specification and parsers with an appropriate set of `drake:`-namespace-prefixed elements and attributes such that material properties can be detailed *with* the model in a robust, supportable way.

## Deliverables:

1. Design document (covering OpenGL-ish Phong/Blinn materials, PBR materials, and *full* BRDF-style materials).
2. Implementation plan for the document.
3. Initial implementation.

SeanCurtis-TRI, 2021-12-04:
The details here should be rolled into #16043 and then this closed.

jwnimmer-tri, 2024-11-01:
Closing as "not planned".  We have evolved a lot since this was written.

Possibly we still desire to have improved parser features, but that should be filed as a new ticket, with new specifics.

closed 2024-11-01 by jwnimmer-tri