RobotLocomotion/drake #11748: [render] Assess IPC render engine implementations

SeanCurtis-TRI, 2019-06-28:
## Problem statement

There are mature renderers (in both render engines and authoring packages) which provide APIs such that we could create a Drake `RenderEngine` that invokes a renderer vis inter-process communication. This could provide a path to exercising mature technologies without trying to embed them into Drake directly.

Two possibilities:
  - Driving uses Unity
  - @gizatt had done a prototype with Blender.

This approach has all sorts of inherent challenges. As the investigation begins, this statement will be extended to offer a more complete discussion of the unique challenges.

EricCousineau-TRI, 2020-01-17:
Thanks! And just to check, do you mean Blender, not Bullet?

sherm1, 2020-01-17:
Oops, yes -- fixed.

jwnimmer-tri, 2021-11-09:
I wonder if this issue should be closed in lieu of #15915, which I believe is the same feature request, but the one actually being worked / tracked?

SeanCurtis-TRI, 2021-12-04:
This is actively being pursued and doesn't need to be tracked here.

closed 2021-12-04 by SeanCurtis-TRI